node-hid/node-hid #244: Failing to access Dymo Scale M10

Arsach, 2018-01-30:
Hello,
Thanks for this great contribution.

I am currently trying to access a Dymo scale through node-hid.
I am running Raspbian GNU/Linux 8.0 (jessie) and I have connected a Dymo M10 scale to USB ports.
l am assuming is it recognized as HID since libusb gives me this:
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     221
         Report Descriptors:

However I am not able to see the scale in the devices list.
Any thought?

Thanks.

todbot, 2018-01-30:
Hi @Arsach,
Are you running as the `root` user or using udev rules for the device (https://github.com/node-hid/node-hid#udev-device-permissions )
You will need to do either to access the device.

Arsach, 2018-02-05:
Hello,
Yes I created the rule for the device. Checked it again but still not able to read anything.

todbot, 2018-02-05:
Hi, after updating the rules and running udevadm, you will need to unplug and replug in the device so the rules can be activatd.

closed 2019-11-23 by todbot