node-hid/node-hid #245: Option to use HidD_SetOutputReport for writing output (DS4)

Pecacheu, 2018-02-03:
Making a new issue for this since #184 is closed. Besides, it's a different problem at this point anyway.

Looking at the [DS4Windows ](https://github.com/Jays2Kings/DS4Windows) source code, specifically the source code of the native HIDAPI wrapper, it has to be down to the difference between **WriteOutputReportViaControl** and **WriteOutputReportViaInterrupt**.

In [DS4Device.cs](https://github.com/Jays2Kings/DS4Windows/blob/a06d77641febd57a7f5934d039dd81f9838306cb/DS4Windows/DS4Library/DS4Device.cs#L288), WriteOutputReportViaControl is called when using bluetooth, which calls the function in [HidDevice.cs](https://github.com/Jays2Kings/DS4Windows/blob/a06d77641febd57a7f5934d039dd81f9838306cb/DS4Windows/HidLibrary/HidDevice.cs#L251). WriteOutputReportViaControl uses the **HidD_SetOutputReport** function, while WriteOutputReportViaInterrupt writes to the device using a filestream.

Meanwhile, node-hid uses only the filestream (aka WriteOutputReportViaInterrupt) approach for `.write`
I tried changing this to HidD_SetOutputReport in `hidapi\windows\hid.c` and recompiling using node-gyp. The file compiled successfully with no errors, however the program crashes silently when attempting to call `.write`. The program must be crashing during the call to **HidD_SetOutputReport**, as I tried intentionally crashing the program on the next line in **hid.c**, and the program crashes before this.

```
int HID_API_EXPORT HID_API_CALL hid_write(hid_device *dev, const unsigned char *data, size_t length) {
	if(TRUE) { //WriteOutputReportViaControl
		BOOL res = HidD_SetOutputReport(dev->device_handle, (PVOID)data, length);
		if(!res) { register_error(dev, "HidD_SetOutputReport"); return -1; }
		return length;
	} else { //WriteOutputReportViaInterrupt
		/* ... Normal node-hid function ... */
```

EDIT: Recompiled DS4Windows to use WriteOutputReportViaInterrupt over bluetooth and application behaved normally except LED/rumble settings had no effect. So that's definitely why it's not working in node-hid. (I propose a second boolean argument to `.write`)

todbot, 2018-02-03:
Hi,
I'm a bit unclear what you're asking for.  The node-hid `device.write()` is doing a HID Output Report using standard `hidapi` that's been shown to work across multiple OSes and multiple devices.

There is a known Windows-specific issue with `device.write()` where you need to prepend the reportId to the buffer, even when the device's HID descriptor doesn't use reportIds. See https://github.com/node-hid/node-hid#prepend-byte-to-hid_write

If that does not solve your issue, it sounds like what you want is a change to upstream `hidapi`, but I'm still not sure what.  Do you want to use `HidD_SetOutputReport` instead of `WriteFile()`? If so, it looks like this is similar to: https://github.com/signal11/hidapi/issues/218 ?  If so, that is again outside of `node-hid`'s purview.

Pecacheu, 2018-02-03:
Okay thanks.

closed 2018-02-03 by Pecacheu

Pecacheu, 2018-02-15:
## Update:
So after endlessly searching the [hidapi fork tree](https://github.com/signal11/hidapi/network) for hours, I found this one: https://github.com/jarveson/hidapi
It successfully implements what I was trying to do with `hid.c` in the code above!

I've created a fork of node-hid here: https://github.com/Pecacheu/node-hid that accepts an optional boolean argument to `hid.write` which can be set to true to use hid_write_control instead.

With this change, now the DS4 can work via bluetooth!
https://github.com/Pecacheu/dualshock

I don't know if I can make a fork request on node-hid, though, since it also uses a fork of hidapi.