node-hid/node-hid #246: Version `GLIBCXX_3.4.21' Not Found

DGruenwald, 2018-02-07:
Hi. I'm trying to use node-hid with electron (`electron --version` gives me v1.7.12) and I've followed your instructions in the readme. However, when I try to use it, I get the following error:
```Uncaught Exception:
Error: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /home/-project path-/node_modules/node-hid/build/Release/HID.node)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
...
```
If I install node-hid with `--build-from-source` it works without any problems in standalone node, but then I can't use it with electron since I'll get a "compiled against a different Node.js version" error.
And running `electron-rebuild` just get's me back to the "GLIBCXX_3.4.21" error above.

I'm running Linux Mint 17 Cinnamon 64-bit, if that helps.

todbot, 2018-02-07:
Hi,
Do you have a link to your project?  What does your project's code and pacakge.json look like?

DGruenwald, 2018-02-07:
It's not much of a project yet. All I've done so far is trying to install electron and node-hid. The package.json is just the standard `npm init -y` but with 
```json
"scripts": {
	"start": "./node_modules/.bin/electron .",
	"postinstall": "electron-rebuild --force"
},
"devDependencies": {
	"electron": "1.7.12",
	"electron-rebuild": "1.7.3"
},
"dependencies": {
	"node-hid": "0.7.2"
}
```
and the index.js:
```javascript
var hid = require("node-hid");
var devices = hid.devices();
console.dir(devices);
```

todbot, 2018-02-08:
Hi, 
Short answer: use `node-hid@0.5.7`

I installed Linux Mint 17 to try this out and was able to replicate what you're seeing.
I am using the https://github.com/todbot/electron-hid-toy project.

The crux of the problem appears to be that the `node-hid` prebuilt binaries for Electron 1.7 are built on a newer version of libstdc++ than ships with Mint 17.  And there appears to be no way of upgrading libstdc++ without upgrading to Mint 18.

Since `node-hid@0.5.7` prebuilt binaries for Electron 1.7, you can use it to build from source.

##### Notes for future me and others looking to work on this issue:

To get a complete development environment in Mint 17, install the following:
```
sudo apt install build-essential
sudo apt install libusb-1.0-0 libusb-1.0-0-dev libudev-dev
```
This will allow `npm install node-hid --build-from-source` to build completely and run in the Node command-line.

However when doing `electron-rebuild` with `node-hid@0.7.2`, the `prebuild-install` module is correctly pulling down a "Linux build for Electron 1.7.x", but there's no way to force `prebuild-install` to `--build-from-source` when being called from `electron-rebuild`.

Going forward, it may be that the correct thing to do is create `node-hid` prebuilt binaries on Ubuntu 16.04 LTS instead of Ubuntu 17.

Also, keep poking at `prebuild-install` and `electron-rebuild` to see if it's possible to force a `--build-from source`

Keeping this issue open for now until I investigate Ubuntu 16 libstdc++.

DGruenwald, 2018-02-09:
Digging through the source of electron-rebuild, I've found that it checks package.json-files for the "prebuild-install" dependency to determine whether to build from source, so I've temporarily removed it from node-hid's package.json in my local node_modules directory, which seems to have done the trick.

It's not really a proper solution, but now I can at least start developing...

todbot, 2018-02-09:
Hi @DGruenwald,
That will totally work but if you want to have your app not need that hand editing, you can put "node@0.5.7" in your package.json dependencies.  

There's no functional difference between 0.5.7 and 0.7.2 for Linux users beyond the `prebuild` stuff (which isn't working for you because of the libc stuff)

DGruenwald, 2018-02-09:
> There's no functional difference between 0.5.7 and 0.7.2 for Linux

Correct me if I'm wrong, but wasn't 'hidraw' added later than 0.5.7? At the very least there was no `setDriverType()` function...
It looks like using 'libusb' means some properties like 'usagePage', 'manufacturer' or 'product' are not returned by `devices()`, which is information that I need, though.

todbot, 2018-02-09:
oh yes, my mistake.  That is a big change.  Apologies.  If you need the extra usagePage stuff then hidraw is for you and that means 0.7.2. 

The prebuild stuff has caused a lot of issues. Good ones, as the alternative was much worse, but some things like `setDriverType()` may not get solved any time soon.

I'm still looking into ways to short-circuit `prebuild-install` so it'll always build.  I've not found it yet so I think I'll open an issue on it.

todbot, 2018-02-09:
okay here's one solution so you don't have to hack node-hid's package.json:
In your app's package.json, make your node-hid dependency be:
```
    "node-hid": "github:node-hid/node-hid#v0.7.2",
```
This appears to short-circuit `prebuild-installs` checks and forces compilation.

todbot, 2018-02-10:
I rebuilt the Linux prebuilds using an older libc and and re-uploaded them.  The electron-hid-toy app now rebuilds and works on Linux Mint 17. (and Ubuntu 16, which also had this problem)

(Note you'll need to `rm -rf ~/.npm /_prebuilds` to remove the cache of past builds)
Thanks for pointing the original issue.

closed 2019-06-19 by todbot