node-hid/node-hid #248: Can't Open PS4 Controller on Mac

UsersDoProgram, 2018-03-20:
Every time I execute the following code on Mac OS X to try and open my PS4 Controller as an HID device an error is raised. I have tried logging into root and sudo, and it doesn't yield a different result. How could I fix this?
`const HID = require("node-hid");`
`const hidDev = HID.HID(0x05c4,0x054c);`

/node_modules/node-hid/nodehid.js:54
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^
Error: cannot open device with vendor id 0x5c4 and product id 0x54c
    at Object.HID

todbot, 2018-03-20:
Do you have any other program accessing it at the same time?
Is this over USB or Bluetooth?  With Bluetooth you may have issues.
If USB, does it show up in the device list in System Information -> USB?

UsersDoProgram, 2018-03-21:
No, its just the one program I made that tries to claim the device from the OS. When I plug the controller in via USB and Check the information it does show up. The only thing I could think of is that my Mac has USB C ports and I am connecting the controller through a USB C to 3.0 adapter.

obiwanjacobi, 2018-11-10:
I have a similar problem. The USB device shows up in System Information - but under a (VM) hub- but HID reports the same error (different vendor and product ids). Then I tried a Java program that accesses the same device and that works fine. My code was developed under Windows and should be solid (its not a bug in my code).

todbot, 2019-11-23:
There are a few DualShock4 Node libraries to crib from. I would use one of those. 
https://www.npmjs.com/package/dualshock
https://www.npmjs.com/package/ds4

closed 2019-11-23 by todbot