node-hid/node-hid #251: Package Electron App with node-hid for Windows

stevanmilic, 2018-03-27:
What is the correct way to build a Electron App for Windows with node-hid as one of the dependencies?

To be more precise, I'm building electron app from Linux, where in `postinstall` script I'm having `electron-rebuild`, where my target architecture is win32-x64.

For my Linux env everything works fine. But as soon as I try running my application on windows I get the fallowing error: 

```
ERROR Error: Uncaught (in promise): Error: %1 is not a valid Win32 application.
\\?\C:\Users\steva\AppData\Local\Temp\35FC.tmp.node
Error: %1 is not a valid Win32 application.
```

So, my thinking is next, I'm having this problem since I'm rebuilding node-hid for specific electron node version, and that's where the incompatibility comes from. But sincerely I'm not sure.

Should I build my app from Windows? Is it even possible to do it from Linux?

Thanks guys!

stevanmilic, 2018-03-28:
I suspect this is because I use version `0.4.0` of node-hid, _or_ this is what third party library uses.
I'll try to use the new version, I see that that one uses prebuilt binaries.

stevanmilic, 2018-03-28:
It looks like `prebuild-install` installs linux prebuild, although different platform is specified from `electron-packaged`. Any help, how to specify the windows prebuilt version?

todbot, 2018-03-28:
Are you talking about `node-hid@0.4.0`? I don't think there was a prebuild system in place then.  The current prebuild system works well in Windows, Mac and Ubuntu Linux.  What issue are you seeing exactly? (log files, error messages, etc)

todbot, 2018-03-28:
As for how to use node-hid with Electron, please check out: 
https://github.com/node-hid/node-hid/#using-node-hid-in-electron-projects
For examples, you can see:
https://github.com/todbot/electron-hid-toy
and
https://github.com/todbot/Blink1Control2

stevanmilic, 2018-03-28:
Yea I switched to `node-hid@0.7.2`, but it looks like `prebuild-install` doesn't install the right platform version https://github.com/prebuild/prebuild-install/blob/master/rc.js#L34, somehow it is stuck my the OS one, and not reading it from `electron-packager`

todbot, 2018-03-28:
I'm sorry I don't understand, how is that line in prebuild the problem?  (and if it is, this sounds like a bug in prebuild and others would see it)

stevanmilic, 2018-03-28:
Okay, this is definitely not a problem with `node-hid`, sorry for disturbance, so I'm closing this.

The error occurs when you start packaged(i.e. build for production) Electron application from Windows.
`Error: %1 is not a valid Win32 application.`

Since, I'm building an Electron app for Windows from Linux, I'm executing the following command:
`electron-packager --arch=x64 --platform=win32 ...`
As you see I specified _win32_ as platform, but the problem comes when we try to install `node-hid` with `prebuild-install`, then that module doesn't read _win32_ as platform, and uses _linux_ as platform variable. And as consequence it downloads wrong prebuild version of `node-hid`, i.e. it install _linux_ prebuild version instead of _win32_.

The problems can be solved if you put in `.npmrc` file `platform=win32`, and in that way override the platform version.

closed 2018-03-28 by stevanmilic

todbot, 2018-03-28:
Oh you're building a Windows Electron app on Linux? Yes you will have many problems there.  While it is possible for some simple things, I have had nothing but bad experiences trying to build anything of substance that way. I've taken to doing Windows builds using Appveyor or a Windows 10 VM.  Both work pretty well, if you can stomach PowerShell.

stevanmilic, 2018-03-28:
Yea, that's what I'm doing. It worked fine until I started using native node app(`node-hid`).
It looks like I'll have to build it on Windows after-all...

todbot, 2018-03-28:
It's super irritating. I feel your pain.  I also had troubles with (I think) an XML parser, a config library, and an MQTT library.

stevanmilic, 2018-03-28:
Yea, very very irritating. After I build it from Windows, my app works fine. Thanks!