node-hid/node-hid #254: Electron App Crash at startup on Win10

theCrius, 2018-04-17:
#### Issue Details
I'm trying to use node-hid on electron but get a crash report windows as soon as the window open up.
The error reported is the following:

[image]

#### Code sample
If needed I can provide a github with the code as it's in early stage and there isn't much done, but the code itself is fairly simple.

This is `main.js`
```javascript
const { app, BrowserWindow } = require('electron');
const HID = require('node-hid');

// Process Management
const shouldQuit = makeSingleInstance();
if (shouldQuit) return app.quit();

app.on('ready', () => {
  createWindow();
  mainWindow.webContents.on('did-finish-load', () => {
    mainWindow.webContents.send('usbDevices', HID.devices());
  });
});

app.on('activate', () => {
  if (mainWindow === null) {
    createWindow();
  }
});

// Windows Management
let mainWindow = null;

function createWindow(){
  mainWindow = new BrowserWindow({
    width: 800,
    height: 600,
    backgroundColor: '#ffffff',
    icon: `file://${__dirname}/dist/assets/scales.png`
  });
  mainWindow.loadURL(`file://${__dirname}/dist/index.html`);

  // EVENTS
  mainWindow.on('closed', () => {
    mainWindow = null;
  });
}

// Utilities

function makeSingleInstance () {
  return app.makeSingleInstance(() => {
    if (mainWindow) {
      if (mainWindow.isMinimized()) mainWindow.restore();
      mainWindow.focus();
    }
  });
}
```

And the webapp code is built on angular (not angularJS) with everything coming out of a clean `ng new` project except the content of the app.component that is:

```typescript
import { Component, OnInit } from '@angular/core';
import { ipcRenderer } from "electron";

@Component({
  selector: 'app-root',
  templateUrl: './app.component.html',
  styleUrls: ['./app.component.css']
})

export class AppComponent implements OnInit {
  title = 'app';

  constructor() { }

  getUsbDevices() {
    console.log("Hey");
    ipcRenderer.on('usbDevices', (event, message) => {
      console.log(message);
    });
  }

  ngOnInit() {
    console.log('Starting App');
    this.getUsbDevices();
  }
}
```

Just to be sure it wasn't an issue with the code in angular, I got rid of any electron usage on the webapp and it still output the error.

I'm fairly new to electron and most of all, working with USB-related stuff so i could be doing something very stupid and not noticing it.

#### Versions Info:
Electron Version: 1.7.13 (Downgraded from 1.8.4, latest stable that had the exact same issue)
node-hid: 0.7.2
electron-rebuild: 1.7.3
Windows Pro: 16299.248

Full package JSON: https://gist.github.com/theCrius/d0793ca3699aa729c4db9bc188d33ad3

todbot, 2018-04-17:
That looks like the kind of error you'd see when the native library part of `node-hid` cannot be found.  Common reasons for this are:
- `electron-rebuild` is not being run.  `electron-rebuild` recompiles the native library part of node-hid (or fetches if there's a prebuilt one for your combo of OS + Electron version)
- Electron `node_modules` path is moved/changed 

I suspect the problem is the latter.   I can't recreate your project because `npm run start` errors out for me due to some Angular config file error, but I bet Angular is packing/moving the .js and not accounting for the native modules.  I've usually solved this with a webpack or similar config change.

I know nothing about how Angular works, but it looks like these two projects:
https://github.com/meltedspark/electron-angular-native
https://github.com/maximegris/angular-electron
have some info on how to do Angular in Electron with native libraries.

theCrius, 2018-04-17:
@todbot I never thought about what angular does with the node_modules and you could honestly be onto something.

I'll check out the two repo you linked, tomorrow and update with whatever I discover or if one of the two "just works" :)

Anyway I tried to run my repo on a linux machine I had around and it works flawlessly there. 

Unfortunately I have to develop this on the company machine that has windows. I'm afraid the issue is somewhere around the matrix of builds that are needed by my version of Windows + Node 8.x + Electron 1.7.x + node-hid.

Unfortunately I'm not well versed in the underlying technologies of those software so I cannot really make an educated guess.

theCrius, 2018-04-18:
Ok then. I went on and started a new project out of [AngularElectron ](https://github.com/maximegris/angular-electron)

On this window machine it took a couple more things to make it work ([check here](https://github.com/maximegris/angular-electron/issues/172)) but in the end it started, no errors, all good.

I forked it and pushed the changes I made as well: https://github.com/theCrius/angular-electron-from-maximegris

All I did was:

`npm i --save node-hid`

Then I added these lines in the main.ts (that will be compiled in main.js)

```javascript
import { ipcMain } from 'electron';

//[...]
// Listen for async message from renderer process
ipcMain.on('getUsbDevices', (event, arg) => {
  event.sender.send('usbDevices', 'Sending USB Devices Found');
  event.sender.send('usbDevices', HID.devices());
});
//[...]
```

I also added in the `src/app/app.component.ts` these lines:

```javascript
import { OnInit } from '@angular/core';

//[...]
export class AppComponent implements OnInit { //added 'implements OnInit'
//[...]

getUsbDevices() {
    this.electronService.ipcRenderer.send('getUsbDevices', true);
    this.electronService.ipcRenderer.on('usbDevices', (event, message) => {
      console.log(message);
    });
  }

  ngOnInit() {
    this.getUsbDevices();
  }
```

To whoever this would not be clear, what should happens is that Electron start and wait, listening for a ping from the app.
The app start and once is ready to render the content, ping electron and ask for the USB devices.

You can see the exact commit here:
https://github.com/theCrius/angular-electron-from-maximegris/commit/3c6828eb92c320e849d9fc9f9542aa8000c0af08

Now at that point the same error comes back.

[image]

if you just comment the line that invoke HID.devices(), there is no message and everything works fine.
I'll try running again this on my personal computer that has Ubuntu 16.04 instead of the office's one but I'm afraid I already know the result.

Something is missing for windows and I cannot guess what it is.

theCrius, 2018-04-18:
I just realised I forgot to downgrade to Electron 1.7.13!

And it works it seems 👍 

[image]

closed 2018-04-18 by theCrius