node-hid/node-hid #255: More than one event when data is more than 64 characters

ndrantotiana, 2018-04-17:
I use "node-hid" to scan data from an AZTEC 2D-barcode. The data contained in the image is more than 64 characters in length and I received more than one "data" event. Is it normal or is it a misconfiguration of my HID scanner? If it is normal, anyone could help me on how to check if data should be concatenated until the last "data" event is occurred or if it is a new data from a new barcode scanning?
Thank you in advance.

todbot, 2018-04-17:
Hi,

Can you post the HID descriptor for the barcode scanner and a code snippet showing how you're reading from it?

It maybe be the HID scanner is set up to send data in 64-byte chunks and thus to send more than that, it will do multiple transfers.

ndrantotiana, 2018-04-17:
How to get the HID descriptor please?

todbot, 2018-04-17:
Hi,
If you're on Windows, you can use TDD at http://www.thesycon.de/eng/usb_descriptordumper.shtml 
If you're on Mac OS X, you can use USB Prober, available by following the steps outlined here: https://superuser.com/a/814339

ndrantotiana, 2018-04-17:
Sorry, I had to do it on my friend's Windows and it is in french. This is what I get from USB Descriptor Dumper:

> Information

 for device Symbol Bar Code Scanner (VID=0x05E0 PID=0x1300):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x001F
Current configuration value: 0x00
Number of open pipes: 0

Device Descriptor:
------------------------------
0x12    bLength
0x01    bDescriptorType
0x0200    bcdUSB
0x00    bDeviceClass      
0x00    bDeviceSubClass  
0x00    bDeviceProtocol  
0x10    bMaxPacketSize0   (16 bytes)
0x05E0    idVendor
0x1300    idProduct
0x0100    bcdDevice
0x01    iManufacturer   "Symbol Technologies, Inc, 2008"
0x02    iProduct   "Symbol Bar Code Scanner"
0x03    iSerialNumber   "S/N:C2F1944B2C545D4981E58B910913E5DB Rev:PAABJS00-005-R02-8"
0x01    bNumConfigurations

Configuration Descriptor:
------------------------------
0x09    bLength
0x02    bDescriptorType
0x0022    wTotalLength   (34 bytes)
0x01    bNumInterfaces
0x01    bConfigurationValue
0x08    iConfiguration   "Bus Powered"
0xA0    bmAttributes   (Bus-powered Device, Remote-Wakeup)
0xFA    bMaxPower      (500 mA)

Interface Descriptor:
------------------------------
0x09    bLength
0x04    bDescriptorType
0x00    bInterfaceNumber
0x00    bAlternateSetting
0x01    bNumEndPoints
0x03    bInterfaceClass      (Human Interface Device Class)
0x00    bInterfaceSubClass  
0x00    bInterfaceProtocol  
0x05    iInterface   "Hand-Held Scanner (Usage = 4B00h, Usage Page = FF45h)"

HID Descriptor:
------------------------------
0x09    bLength
0x21    bDescriptorType
0x0110    bcdHID
0x00    bCountryCode
0x01    bNumDescriptors
0x22    bDescriptorType   (Report descriptor)
0x001E    bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07    bLength
0x05    bDescriptorType
0x81    bEndpointAddress  (IN endpoint 1)
0x03    bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040    wMaxPacketSize    (1 x 64 bytes)
0x03    bInterval         (3 frames)

Microsoft OS Descriptor:
------------------------------
0x02    bLength
0x03    bDescriptorType
Hex dump: 
0x02 0x03

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Symbol Technologies, Inc, 2008"
0x02   0x0409  "Symbol Bar Code Scanner"
0x03   0x0409  "S/N:C2F1944B2C545D4981E58B910913E5DB Rev:PAABJS00-005-R02-8"
0x08   0x0409  "Bus Powered"
0x05   0x0409  "Hand-Held Scanner (Usage = 4B00h, Usage Page = FF45h)"

------------------------------

Connection path for device: 
Contrôleur d’hôte compatible xHCI USB
Root Hub
Symbol Bar Code Scanner (VID=0x05E0 PID=0x1300) Port: 2

Running on: Windows 10 or greater

Brought to you by TDD v2.11.0, Mar 26 2018, 09:54:50

todbot, 2018-04-17:
I was wrong and TDD doesn't give the HID descriptor, but it does give the config descriptor, which in this case I think is fine.  It looks like that device does send data in 64-byte packets.

So I believe you'll receive multiple `data` events if the data is more than 64-bytes.

ndrantotiana, 2018-04-17:
Di you see it using the following?

> 0x0040 wMaxPacketSize (1 x 64 bytes)

So it's an issue of my barcode scanner not in node-hid?  By chance, do you know what should I do to increase it?

ndrantotiana, 2018-04-17:
Do you know how to delimitate the beginning and the end of the data from the scanner or is it scanner specific? Sorry for all my questions.

DeekshithShetty, 2018-07-10:
@ndrantotiana I believe the scanner needs to send a packet telling how many 64 byte packets of data is going to come. Using this information you can append data until the no of packets is received.

todbot, 2019-06-19:
From the `wMaxPacketSize (1 x 64 bytes)` in the descriptor referenced, that means you will receive multiple reports if the data payload being sent is >64 bytes.  That's just how USB works.

closed 2019-06-19 by todbot