node-hid/node-hid #258: Interface 0 disappeared with node-hid 0.7.2

mondalaci, 2018-05-05:
Trying to upgrade https://github.com/UltimateHackingKeyboard/agent from node-hid 0.5.7 to 0.7.2 and it has proven to be more problematic than anticipated. Interface 0 was seen by 0.5.7 but not by 0.7.2.

This is the interface list with 0.5.7:

```
[ { vendorId: 7504,
    productId: 24866,
    path: '0002:000f:00',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 0 },
  { vendorId: 7504,
    productId: 24866,
    path: '0002:000f:01',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 1 },
  { vendorId: 7504,
    productId: 24866,
    path: '0002:000f:02',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 2 },
  { vendorId: 7504,
    productId: 24866,
    path: '0002:000f:03',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 3 },
  { vendorId: 7504,
    productId: 24866,
    path: '0002:000f:04',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 4 }]
```

And this is with 0.7.2:

```
[  { vendorId: 7504,
    productId: 24866,
    path: '/dev/hidraw8',
    serialNumber: '',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 1 },
  { vendorId: 7504,
    productId: 24866,
    path: '/dev/hidraw9',
    serialNumber: '',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 2,
    usagePage: 25209,
    usage: 24943 },
  { vendorId: 7504,
    productId: 24866,
    path: '/dev/hidraw10',
    serialNumber: '',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 3,
    usagePage: 25209,
    usage: 24943 },
  { vendorId: 7504,
    productId: 24866,
    path: '/dev/hidraw11',
    serialNumber: '',
    manufacturer: 'Ultimate Gadget Laboratories',
    product: 'Ultimate Hacking Keyboard',
    release: 257,
    interface: 4,
    usagePage: 25209,
    usage: 24943 } ]
```

Any help would be much appreciated!

todbot, 2018-05-06:
Hi @mondalaci,
The main change is that by default the Linux version of `node-hid` uses the OS's "hidraw" driver instead "libusb".  So it sounds like interface 0 isn't actually a HID device or for some reason hidraw is excluding it?

This is something `node-hid` (or really, `hidapi` the C-library it wraps) cannot control.  However, you can force `node-hid` to use the "libusb" version of `hidapi` instead of the "hidraw" default with:

```
npm install node-hid --build-from-source --driver=libusb
```

mondalaci, 2018-10-23:
@todbot Thanks for your insight! In the meantime, we've delved more into this issue, and can provide a much more precise report.

Please see the https://github.com/ert78gb/node-hid-reconnect-problem repo. This repo contains a very simple node-hid based script. The old directory uses node-hid 0.6.0 and the new directory uses node-hid 0.7.0.

I can connect our keyboard, run the new script any number of times, and get:

```
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw2","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":0,"usagePage":11825,"usage":12080}
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw3","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":1,"usagePage":60304,"usage":572}
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw4","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":2,"usagePage":18432,"usage":17481}
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw5","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":3,"usagePage":17457,"usage":12341}
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw6","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":4,"usagePage":6976,"usage":574}
```

Then I can run the old script any number of times, and get the same result.

However, if I run the new script after running the old script then interface 0 disappears until I reconnect the USB device:

```
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw3","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":1,"usagePage":64}
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw4","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":2,"usagePage":18432,"usage":17481}
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw5","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":3,"usagePage":11828,"usage":12084}
{"vendorId":7504,"productId":24866,"path":"/dev/hidraw6","serialNumber":"","manufacturer":"Ultimate Gadget Laboratories","product":"Ultimate Hacking Keyboard","release":257,"interface":4,"usagePage":25209,"usage":24943}
device info: undefined
/home/laci/projects/node-hid-reconnect-problem/new/index.js:13
const device = new HID.HID(deviceInfo.path);
                                      ^
TypeError: Cannot read property 'path' of undefined
```

In other words, node-hid<=0.6.0 makes interface 0 invisible for node-hid>=0.7.0.

Then I tried to build the new script with `npm install node-hid --build-from-source --driver=libusb` and this way the new script worked even after the old script.

1. According to the above, the libusb and hidraw drivers don't play nicely together. Is this a known issue?
2. Is there a known solution?
3. What's the advantage of the hidraw driver over libusb?

todbot, 2018-10-23:
In general with Linux, `hidraw` seems to be the better way to access HID devices as it utilizes the OS's native HID functionality that extends to Bluetooth as well as USB devices. The underlying API in hidraw is also much simpler and more uniform, making it possible to write a HID client that works across more distros of Linux without recompilation.

The `libusb` driver uses low-level USB calls to directly talk to HID devices, bypassing the OS's HID layer.  This means no Bluetooth devices and a more complicated, more brittle HID api that often requires recompiling for different distros. 

The  downside of `hidraw` I think you are seeing is that since this HID device is a keyboard, the OS is claiming it and removing it from possible hidraw devices.  There are ways around this, but then you lose the keyboard as an input device.  On Windows and Mac OS X, one also cannot access HID keyboards because of the OS's claiming of the device. 

If you're making an app for a keyboard and have control of the entire software stack, you should probably be using the `libusb` driver for `node-hid`.  I did try having a run-time switch for driver type, but it wasn't compatible with the `prebuild` system used to create pre-built binaries that many users need.

todbot, 2019-06-19:
Both `hidraw` and `libusb` driver types are now supported and you can set which one you want when first starting the library with `HID.setDriverType()`.

closed 2019-06-19 by todbot