node-hid/node-hid #261: device.write() hangs on macOS

gniezen, 2018-05-30:
Using node-hid 0.7.2 in an Electron app on macOS El Capitan:

- On a specific machine, sending a message using `device.write()` completely hangs the app, such that the timeout event doesn't even fire. 
- This is reproducible, but only on a specific machine (thus far). 
- The same app successfully communicates to the device on other machines (El Capitan, High Sierra and Windows 10).
- Other software can communicate with the device on the same machine.

Even if it is somehow due to the device itself being defective, I was wondering if it's possible to always have `device.write()` return, so that we can at least timeout and display an error message?

Is there any way to debug this further?

todbot, 2018-05-30:
Hi @gniezen,
This sounds like a hardware problem. I would use the Console app to look for any sort of kernel or USB error messages around the time of doing `device.write()`.  `node-hid` is a real thin wrapper on `hidapi`, so if it's hanging the process, it's hanging in native code doing standard USB calls.

The first thing I would try when debugging this would be to make all aspects of the hardware the same as possible between systems: same device, same USB cable, same USB hubs , etc.  I've had USB cables that work intermittently on some machines but well on others (presumably due to worn USB ports on the host)

Finally, is the hanging machine using a USB 3.0 port and the working machines have a USB 2.0 port?  Some implementations of USB firmware (notably the software V-USB stack) have a hard time with USB 3.0.  A fix there is to insert a USB 2.0 hub in between the USB 3.0 host and the the device.

gniezen, 2018-05-31:
Things we checked:

- [x] Same device
- [x] Same USB cable (this is a cradle, with the USB chip inside the connector which plugs into the USB port)
- [x] Tried different USB ports on the same machine
- [x] No USB hub used
- [x] system settings show that it's plugged into a USB 2.0 bus

Now, this is super weird: Following your suggestion of using Console.app I saw a bunch of `com.displaylink.usbnivolistener` error messages repeating every 5 seconds. I doubted that it was related to the issue, but wanted to get rid of the flood of messages so we could see if anything else was happening. We reinstalled (as their installation was corrupted) and then uninstalled the DisplayLink USB drivers, rebooted, and suddenly it worked! Not sure how a USB display driver can affect the USB HID driver, but I'm just glad that after more than a week of struggling with this I can finally put this issue to bed. Thanks @todbot !

closed 2018-07-16 by gniezen