node-hid/node-hid #263: Facing an issue while trying to write from MAC

SravanBalivada, 2018-06-15:
Hi,

I am facing some issues while trying to interact with the device from Mac OS(10.13.4). It is working fine in windows OS but the same code is not working in MAC. I tried all the possible available solutions but none of them seems to work.

core.js:1350 ERROR Error: Uncaught (in promise): Error: Cannot write to HID device
Error: Cannot write to HID device
    at HidUSBService.webpackJsonp../src/app/services/hidUSB.service.ts.HidUSBService.write (hidUSB.service.ts:157)
    at HidUSBService.webpackJsonp../src/app/services/hidUSB.service.ts.HidUSBService.sudoSuper (hidUSB.service.ts:62)
    at HidUSBService.webpackJsonp../src/app/services/hidUSB.service.ts.HidUSBService.process (hidUSB.service.ts:37)

This is the error that I am facing and my code looks like this:

const buffer = new Uint8Array(a);
		const os = 'win32';
		if (os === 'win32') {
			return this.getWindowsSpecStrBuffer(buffer);
		} else {
			return this.getMacSpecStrBuffer(buffer);
		}
	}
	private getWindowsSpecStrBuffer(buffer) {
		const bf = [];
		bf.push(0);
		buffer.forEach(element => {
			bf.push(element);
		});
		return bf;
	}
	private getMacSpecStrBuffer(buffer) {
		const bf = [];
		bf.push();
		buffer.forEach(element => {
			bf.push(element);
		});
		return bf;
	}

I tried changing the bits in bf.push but that did not work.

Please help in solving the issue. I am struggling with this issue since long time.

todbot, 2018-06-15:
What kind of device is it?
Do you know the config descriptor and hid descriptor for this device?

todbot, 2018-06-15:
Also why are you doing `bf.push(0)` on windows but `bf.push()` on Mac?

SravanBalivada, 2018-06-15:
I have no idea about config descriptor and hid descriptor. Can you please let me know where can I find them ?

I am trying to interact with a microphone named MOTIV88+.

and I am using this 
private getMacSpecStrBuffer(buffer) {
const bf = [];
buffer.forEach(element => {
bf.push(element);
});
return bf;
}

I forgot to comment bf.push().

todbot, 2018-06-15:
I see no node-hid calls in any of the code you've posted.  Do you have a link to your entire project on github or similar?

SravanBalivada, 2018-06-15:
this is the write method which I am using : 

write(buffer: any): any {
		// return this.device.write(buffer);
		this.device.write(buffer);
		this.device.write(this.stringToBuffer(''));

		const promise = new Promise((resolve, reject) => {
			this.read((err, data) => {
				if (err) {
					reject(err);
				} else {
					resolve(data.toString('utf8'));
					this.stateChangeSubject.next(data.toString('utf8'));
				}
			});

		});
		return promise;
	}

SravanBalivada, 2018-06-15:
I am using this to connect device: 

this.device = new usb.HID(params.vendorId, params.productId);

and this is the write method that I am using:

write(buffer: any): any {
		// return this.device.write(buffer);
		this.device.write(buffer);
		this.device.write(this.stringToBuffer(''));

		const promise = new Promise((resolve, reject) => {
			this.read((err, data) => {
				if (err) {
					reject(err);
				} else {
					resolve(data.toString('utf8'));
					this.stateChangeSubject.next(data.toString('utf8'));
				}
			});

		});
		return promise;
	}

closed 2018-06-21 by SravanBalivada