node-hid/node-hid #266: node-hid does not compile with node v10.x

todbot, 2018-06-26:
Looks like the deprecated NAN APIs have been removed.
The errors (on Mac OS X, similar on other platforms):
```
1 warning generated.
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc:207:45: error: no matching member function for call to 'NewInstance'
    Local<Object> buf = nodeBufConstructor->NewInstance(1, nodeBufferArgs);
                        ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/Users/tod/.node-gyp/10.5.0/include/node/v8.h:3914:44: note: candidate function not viable: requires single argument 'context', but 2 arguments were provided
  V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                           ^
/Users/tod/.node-gyp/10.5.0/include/node/v8.h:3911:44: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                           ^
../src/HID.cc:231:20: warning: 'Call' is deprecated [-Wdeprecated-declarations]
  iocb->_callback->Call(2, argv);
                   ^
../../nan/nan.h:1617:3: note: 'Call' has been explicitly marked deprecated here
  NAN_DEPRECATED inline v8::Local<v8::Value>
  ^
../../nan/nan.h:98:40: note: expanded from macro 'NAN_DEPRECATED'
# define NAN_DEPRECATED __attribute__((deprecated))
                                       ^
../src/HID.cc:409:37: warning: 'Value' is deprecated [-Wdeprecated-declarations]
        serialPointer = (wchar_t*) *v8::String::Value(info[2]);
                                    ^
/Users/tod/.node-gyp/10.5.0/include/node/v8.h:2876:5: note: 'Value' has been explicitly marked deprecated here
    V8_DEPRECATED("Use Isolate version", explicit Value(Local<v8::Value> obj));
    ^
/Users/tod/.node-gyp/10.5.0/include/node/v8config.h:327:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
2 warnings and 1 error generated.
make: *** [Release/obj.target/HID/src/HID.o] Error 1
```

ada-lovecraft, 2018-07-26:
@todbot should commit 90c8ba6 fix this issue? or did fixing the `Nan` deprecations unveil a deeper problem?

todbot, 2018-07-26:
Hi @ada-lovecraft, yes the current version of node-hid (0.7.3) should fix this issue.  I've not closed it yet mostly because I forgot. :)

todbot, 2018-07-26:
Yes, apologies for that. I'm a maintainer on `node-blink` too but haven't had a chance to update it.

jongear, 2019-01-05:
# node-hid@0.7.6 failing on Node 10
It appears that `node-hid@0.7.6` is failing with Node 10.

## Setup
╰─ node -v
v10.14.1
╰─ npm -v
6.4.1
Mojave macOS 10.14.2

```bash
─ npm install node-hid

> node-hid@0.4.0 install /Users/jon/Side/personal/dev-setup/node_modules/logitech-dual-action-controller/node_modules/node-hid
> node-gyp rebuild

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
../hidapi/mac/hid.c:255:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
                if (chars_copied == len)
                    ~~~~~~~~~~~~ ^  ~~~
../hidapi/mac/hid.c:295:20: warning: comparison of integers of different signs: 'CFIndex' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
                if (used_buf_len == len)
                    ~~~~~~~~~~~~ ^  ~~~
2 warnings generated.
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
In file included from ../src/HID.cc:35:
In file included from ../../nan/nan.h:111:
../../nan/nan_new.h:29:56: warning: 'ToInteger' is deprecated [-Wdeprecated-declarations]
To<v8::Integer>(v8::Handle<v8::Integer> i) { return i->ToInteger(); }
                                                       ^
/Users/jon/.node-gyp/10.14.1/include/node/v8.h:2456:10: note: 'ToInteger' has been explicitly marked deprecated here
  inline V8_DEPRECATED("Use maybe version",
         ^
/Users/jon/.node-gyp/10.14.1/include/node/v8config.h:327:29: note: expanded from macro 'V8_DEPRECATED'
  declarator __attribute__((deprecated))
                            ^
In file included from ../src/HID.cc:35:
In file included from ../../nan/nan.h:111:
../../nan/nan_new.h:34:56: error: no matching member function for call to 'ToInt32'
To<v8::Int32>(v8::Handle<v8::Integer> i)   { return i->ToInt32(); }
                                                    ~~~^~~~~~~
/Users/jon/.node-gyp/10.14.1/include/node/v8.h:2437:43: note: candidate function not viable: requires single argument 'context', but no arguments were provided
  V8_WARN_UNUSED_RESULT MaybeLocal<Int32> ToInt32(Local<Context> context) const;
                                          ^
/Users/jon/.node-gyp/10.14.1/include/node/v8.h:2450:30: note: candidate function not viable: requires single argument 'isolate', but no arguments were provided
                Local<Int32> ToInt32(Isolate* isolate) const);
                             ^
In file included from ../src/HID.cc:35:
In file included from ../../nan/nan.h:111:
../../nan/nan_new.h:39:65: error: too few arguments to function call, single argument 'context' was not specified
To<v8::Uint32>(v8::Handle<v8::Integer> i)  { return i->ToUint32(); }
                                                    ~~~~~~~~~~~ ^
/Users/jon/.node-gyp/10.14.1/include/node/v8.h:2435:3: note: 'ToUint32' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<Uint32> ToUint32(
  ^
/Users/jon/.node-gyp/10.14.1/include/node/v8config.h:416:31: note: expanded from macro 'V8_WARN_UNUSED_RESULT'
#define V8_WARN_UNUSED_RESULT __attribute__((warn_unused_result))
                              ^
In file included from ../src/HID.cc:35:
In file included from ../../nan/nan.h:111:
In file included from ../../nan/nan_new.h:191:
../../nan/nan_implementation_12_inl.h:49:38: error: too few arguments to function call, expected 2, have 1
  return v8::BooleanObject::New(value).As<v8::BooleanObject>();
         ~~~~~~~~~~~~~~~~~~~~~~      ^
/Users/jon/.node-gyp/10.14.1/include/node/v8.h:5035:3: note: 'New' declared here
  static Local<Value> New(Isolate* isolate, bool value);
  ^
In file included from ../src/HID.cc:35:
In file included from ../../nan/nan.h:111:
[... 218 lines cut ...]
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
gyp ERR! System Darwin 18.2.0
gyp ERR! command "/usr/local/Cellar/node@10/10.14.1/bin/node" "/usr/local/Cellar/node@10/10.14.1/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/jon/Side/personal/dev-setup/node_modules/xpc-connection
gyp ERR! node -v v10.14.1
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok

> node-hid@0.7.6 install /Users/jon/Side/personal/dev-setup/node_modules/node-hid
> prebuild-install || node-gyp rebuild

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: node-hid@0.4.0 (node_modules/logitech-dual-action-controller/node_modules/node-hid):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: node-hid@0.4.0 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ node-hid@0.7.6
added 66 packages from 46 contributors and audited 226 packages in 9.646s
found 2 low severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
```

jongear, 2019-01-05:
it's important to note that running `npm rebuild --build-from-source` does allow `node-hid@0.7.6`

todbot, 2019-01-05:
Yes, it looks like you are installing node-hid@0.4.0 in that error log.

As of node-hid@0.7.6, node-hid should be working find with Node v10.4+.

This issue will be closed.  There are some issues with certain minor versions of Node v10 wrt native modules, from what I've seen.

closed 2019-01-05 by todbot