node-hid/node-hid #268: apple touch bar conflicts with node-hid

lc090d, 2018-07-02:
when using a computer with an apple touch bar, connecting to any device via ```new HID.HID( vid, pid);``` or ```new HID.HID( path);``` instead will connect to the touch bar.

Everything else works normally—list devices, read data from device—but regardless of the path, vid, or pid, it will always connect to the touch bar.

lc090d, 2018-07-02:
I am specifically connecting to device like this:
```
var deviceInfo = devices.find( function(d) {
    var isR1 = d.vendorId === 0x6E7 && d.productId === 0x8065; // specifically look for our osprey UEI remote
    return isR1;
});

if(deviceInfo) {
	var device = new HID.HID( deviceInfo.vendorId, deviceInfo.productId);
}
```

here's the devices List, device I am trying to connect to is "R1"
```[ { vendorId: 1452,
    productId: 591,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14100000/USB2.0 Hub@14100000/AppleUSB20Hub@14100000/AppleUSB20HubPort@14110000/Keyboard Hub@14110000/AppleUSB20KeyboardHub@14110000/AppleUSB20HubPort@14112000/Apple Keyboard@14112000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14112000,0',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Keyboard',
    release: 116,
    interface: -1,
    usagePage: 1,
    usage: 6 },
  { vendorId: 1452,
    productId: 632,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@1E,3/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Actuator/AppleHSSPIHIDDriver',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 2198,
    interface: -1,
    usagePage: 65280,
    usage: 13 },
  { vendorId: 1452,
    productId: 632,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@1E,3/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Keyboard / Boot/AppleHSSPIHIDDriver',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 2198,
    interface: -1,
    usagePage: 1,
    usage: 6 },
  { vendorId: 1452,
    productId: 632,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@1E,3/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Trackpad / Boot/AppleHSSPIHIDDriver',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 2198,
    interface: -1,
    usagePage: 1,
    usage: 2 },
  { vendorId: 1452,
    productId: 632,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@1E,3/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Accelerometer/AppleHSSPIHIDDriver',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 2198,
    interface: -1,
    usagePage: 65280,
    usage: 3 },
  { vendorId: 1452,
    productId: 632,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@1E,3/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Device Management/AppleHSSPIHIDDriver',
    serialNumber: '',
    manufacturer: '',
    product: 'Apple Internal Keyboard / Trackpad',
    release: 2198,
    interface: -1,
    usagePage: 65280,
    usage: 11 },
  { vendorId: 1452,
    productId: 34304,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14200000/iBridge@14200000/IOUSBHostInterface@6/IOUSBHostHIDDevice@14200000,6',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'iBridge',
    release: 257,
    interface: -1,
    usagePage: 32,
    usage: 65 },
  { vendorId: 1767,
    productId: 32869,
    path:
     'IOService:/IOResources/IOHIDResource/IOHIDResourceDeviceUserClient/IOHIDUserDevice',
    serialNumber: '0057005A1147343532333437',
    manufacturer: 'VRC81-19',
    product: 'R1',
    release: 272,
    interface: -1,
    usagePage: 1,
    usage: 6 },
  { vendorId: 1452,
    productId: 34304,
    path:
     'IOService:/IOResources/IOHIDResource/IOHIDResourceDeviceUserClient/IOHIDUserDevice',
    serialNumber: '',
    manufacturer: '',
    product: '',
    release: 0,
    interface: -1,
    usagePage: 1,
    usage: 6 },
  { vendorId: 1452,
    productId: 34304,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14200000/iBridge@14200000/IOUSBHostInterface@2/IOUSBHostHIDDevice@14200000,2',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'iBridge',
    release: 257,
    interface: -1,
    usagePage: 13,
    usage: 5 },
  { vendorId: 1452,
    productId: 591,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS02@14100000/USB2.0 Hub@14100000/AppleUSB20Hub@14100000/AppleUSB20HubPort@14110000/Keyboard Hub@14110000/AppleUSB20KeyboardHub@14110000/AppleUSB20HubPort@14112000/Apple Keyboard@14112000/IOUSBHostInterface@1/IOUSBHostHIDDevice@14112000,1',
    serialNumber: '',
    manufacturer: 'Apple Inc.',
    product: 'Apple Keyboard',
    release: 116,
    interface: -1,
    usagePage: 12,
    usage: 1 } ]```

After some more research, disabling System Integrity Protection solves the issue, but I would rather not have to disable that.

todbot, 2018-07-02:
Do you have a driver installed for this "R1" device?  It may be snatching the device so node-hid can't access it properly.
Because it looks like [hidapi](https://github.com/signal11/hidapi) (the C library node-hid wraps), is not generating a proper path for the R1 device.

lc090d, 2018-07-02:
There's not a driver, it is functioning as a keyboard and mouse (trackpad), so the OS is grabbing it before node-hid can access. Launching it using sudo has worked on multiple osx devices without touchbars, but that same code on a device with a touchbar cannot connect, even when run with root privileges

ChachiQ, 2019-05-08:
similar issue in other projects:
https://github.com/OpenEmu/OpenEmu/issues/2851

todbot, 2019-05-09:
Hi @ChachiQ, I don't believe node-hid suffers the same issue as the OpenEmu one you're describing.  (At least I cannot replicate it by creating a USB device with no Manufacturer String or Product String. node-hid deals with that fine)

And as I review your original issue report @lc090d, I think the issue is that this VRC81-19 R1 device is acting like a keyboard correct?  If so, MacOS will not allow node-hid to access it for security reasons (otherwise one could trivially build keyloggers). And explains why turning off SIP allowed it to work.

Is there a way to make the R1 go into a "HID data" mode so it doesn't report as a keyboard?

Closing this issue for now as it seems to be the standard "cannot access keyboard" type of problem. Feel free to open it if you think that's in error.

closed 2019-05-09 by todbot