node-hid/node-hid #269: Cannot write to HID device

yehudahkay, 2018-07-03:
Hi, thank you for the framework. 

My barcode scanner manual says the following:

ACK/NAK- After transmitting data, the scanner expects either an ACK (acknowledge, 0x06) or NAK (not acknowledge, 0x15) response from the host. When a NAK is received, the scanner transmits the same data again and waits for either an ACK or NAK. After three unsuccessful attempts to send data when NAKs are received, the scanner issues an error indication and discards the data.

I have both of the following on windows but I am getting 'cannot write to HID device':

//1
device.write( [0x06] );
//2
device.sendFeatureReport([0x06]);

Does anyone know how I can do this please?

yehudahkay, 2018-07-03:
I can also set the baud rate (default 9600) parity bit (default none) data bit (default 8) stop bit (default 1)

todbot, 2018-07-03:
Hi,
From that descriptor dump, the device is a CDC device (USB serialport) and not HID.  You should use `node-serialport` instead.

yehudahkay, 2018-07-03:
Thanks! I can change the communication type, here is the output with HID:

Information for device 2D BarCode Scanner (VID=0x27DD PID=0x0103):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0001
Current configuration value: 0x00
Number of open pipes: 0

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x27DD	idVendor
0x0103	idProduct
0x0103	bcdDevice
0x01	iManufacturer   "二维条码枪"
0x02	iProduct   "2D BarCode Scanner"
0x03	iSerialNumber   "2010-02-01"
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xC0	bmAttributes   (Self-powered Device)
0xC8	bMaxPower      (400 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x01	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0110	bcdHID
0x21	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x003C	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "二维条码枪"
0x02   0x0409  "2D BarCode Scanner"
0x03   0x0409  "2010-02-01"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
2D BarCode Scanner (VID=0x27DD PID=0x0103) Port: 6

Running on: Windows 10 or greater

Brought to you by TDD v2.11.0, Mar 26 2018, 09:54:50

yehudahkay, 2018-07-03:
I am able to read data with the library right now using:

device.on("data", function(data) {
          console.log(data)
}

However if I try to write then the data event is no longer triggered or devtools crash.

closed 2019-11-23 by todbot