node-hid/node-hid #271: sendFeatureReport and getFeatureReport slow to error

sebirdman, 2018-07-06:
I've got a keyboard that i'm sending and getting feature reports on very quickly. 

Sometimes while writing or reading, i receive one of these errors:

https://github.com/node-hid/node-hid/blob/master/src/HID.cc#L375
https://github.com/node-hid/node-hid/blob/master/src/HID.cc#L332

the issue is, when i hit one of these errors node seems to completely halt for a moment before i receive the error. This means that the read/writes stop for a longer than desired period of time. 

Any suggestions on how I can improve this?

chris13524, 2018-07-06:
I am also experiencing this issue. Any help would be greatly appreciated.

todbot, 2018-07-06:
Two things to consider:
1. I assume this is on Linux?  Are you removing the keyboard entirely from the Linux kernel HID module?  If not, then the kernel is likely also accessing the device, causing your request to fail
2. USB HID has a max update rate of 1000 Hz (or 1ms per transfer).  If you are requesting faster than this, you won't get anything and depending on the OS and HID driver underneath, you'll probably get an error.

sebirdman, 2018-07-06:
@todbot thanks for the response

I've observed this behavior on mac, linux, and windows. I'm sure that your point #1 is the issue here. I was trying to get around removing the driver and writing my own, but it looks like that might be the only solution if i want to be able to send/get feature reports and type at the same time (reliably)

closed 2019-11-23 by todbot