node-hid/node-hid #274: node-usb to node-hid migration

bladesling, 2018-08-11:
I'm attempting to migrate from node-usb to node-hid as node-usb is not a well up kept library. I was able to do everything I needed with node-usb except deal with the bugs likely from it not being well up kept.

Using node-usb I use something called a control transfer but I'm not sure what that translates to in node-hid. I would assume its a .write or .sendFeatureReport but I can't seem to get it to send the same way. Any help would be appreciated.
```
device.controlTransfer(0xA1, 0x01, 0x346, 0x0, 0x40, (error, data) => {
    console.log(error, data);
});
```

Using monitoring software this is what the message ends up us
```
000024: Control Transfer (UP), 2018-08-11 03:04:33.8226742 +0.0001856. (1. Device: LapCounter) Status: 0x00000000
Pipe Handle: Control Pipe
 46 0A 01                                          F..
Setup Packet
 A1 01 46 03 00 00 41 00                           ¡.F...A.
Recipient: Interface
Request Type: Class
Direction: Device->Host
Request: 0x1 (Unknown)
Value: 0x346
Index: 0x0
Length: 0x41
```

Here is the device's descriptor dump

```
Information for device LapCounter (VID=0x10C4 PID=0x86B9):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0029
Current configuration value: 0x00
Number of open pipes: 0

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x10C4	idVendor
0x86B9	idProduct
0x0100	bcdDevice
0x01	iManufacturer   "Robitronic Electronic"
0x02	iProduct   "LapCounter"
0x03	iSerialNumber   "002FEE20"
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0397	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Robitronic Electronic"
0x02   0x0409  "LapCounter"
0x03   0x0409  "002FEE20"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
LapCounter (VID=0x10C4 PID=0x86B9) Port: 2

Running on: Windows 10 or greater

Brought to you by TDD v2.11.0, Mar 26 2018, 09:54:50
```

todbot, 2018-08-12:
Hi,
Do you have the HID report descriptor for this device? In HID you deal with "reports" (data packets), either as INPUT, OUTPUT, or FEATURE reports.  In `node-hid`, you use `device.write()` to send OUTPUT reports, `device.read()` to read INPUT reports, and `device.sendFeatureReport()` and `device.getFeatureReport()` to deal with FEATURE reports.  

Without knowing the HID reports the device you're dealing with expects, it's difficult to say what to do.

bladesling, 2018-08-28:
How would I go about getting the reports? Any tools I can use to learn the reports? As I said I have a working example using the node-usb library using control transfers. The problem is that it doesn't work on mac because the hid devices are protected and I think I need to use a library specific for HIDs. Or maybe the device just won't function on the mac.

I'm not a usb guy, just trying to support this device on mac. I understand if there is no help. You can close this ticket. Just wish there was a bit more instruction on how to poke USB devices and in what scenarios you may need a kext or something.

closed 2018-09-29 by bladesling