node-hid/node-hid #275: "could not send feature report to device" on Raspberry Pi

danawoodman, 2018-08-12:
I'm attempting to write to a USB Relay using node-hid. Got everything working great on my Macbook Pro but I'm failing with:

```
Error: could not send feature report to device
```

I've follow the instructions about adding a `.rules` file for udev and reconnecting device as well as running as `sudo` (eg `sudo node myscript.js`) but nothing seems to improve the situation.

I've also installed node-hid from source using `npm i node-hid --buld-from-source`.

It appears I can connect to the device (which I assume means the `sudo` bit is working?). Any thoughts as to why it would not send the report on Raspberry PI (NOOBS latest) but work fine on OSX?

Here is the test code I'm using:

```js
const HID = require('node-hid')
const device = new HID.HID(5824, 1503)
device.sendFeatureReport([0xf3])
```

`5824` is the relay's vendor ID, `1503` is the product ID and `0xfe` is the "on" command for the relay (working on OSX). 

Thanks!

todbot, 2018-08-12:
Hi,
So your code works on MacOS but not on Raspberry Pi?  
I would double-check the udev rules.  Could you copy them to this issue so I can check them?

Be sure to unplug your device, add the rule, then run  `udevadm control --reload-rules && udevadm trigger` as root.  Then plug back in your device. You can run `udevadm monitor` to see if the rule is triggered when you plug the device in.  

(Oh and in general, you should not run Node.js things as root. Because of how Node works, 'sudo' will not ensure your code runs with elevated permissions and Node writes temp files that could end up being owned as root, making running things as non-root difficult)

danawoodman, 2018-08-20:
My rules file `/etc/udev/rules.d/doorlock.rules`:

```
SUBSYSTEM=="input", GROUP="input", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="05df", MODE:="666", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="05df", MODE="0666", GROUP="plugdev"
```

Here is the output of `udevadm monitor`:

```
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[181.295902] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2 (usb)
KERNEL[181.296248] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0 (usb)
KERNEL[181.299505] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:16C0:05DF.0004 (hid)
KERNEL[181.299764] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/usbmisc/hiddev1 (usbmisc)
KERNEL[181.300398] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:16C0:05DF.0004/hidraw/hidraw3 (hidraw)
UDEV  [181.323240] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2 (usb)
KERNEL[181.345751] add      /module/media (module)
KERNEL[181.345849] add      /bus/media (bus)
UDEV  [181.348967] add      /module/media (module)
UDEV  [181.349726] add      /bus/media (bus)
KERNEL[181.363560] add      /module/videodev (module)
KERNEL[181.363986] add      /class/video4linux (class)
KERNEL[181.368543] add      /module/radio_ma901 (module)
KERNEL[181.368660] add      /bus/usb/drivers/radio-ma901 (drivers)
UDEV  [181.369687] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0 (usb)
UDEV  [181.371468] add      /class/video4linux (class)
UDEV  [181.372357] add      /module/videodev (module)
UDEV  [181.376041] add      /bus/usb/drivers/radio-ma901 (drivers)
UDEV  [181.377674] add      /module/radio_ma901 (module)
UDEV  [181.378360] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/usbmisc/hiddev1 (usbmisc)
UDEV  [181.379489] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:16C0:05DF.0004 (hid)
UDEV  [181.383544] add      /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:16C0:05DF.0004/hidraw/hidraw3 (hidraw)
```

todbot, 2019-11-23:
I'm going to assume you got this working?  Let me know if not.

closed 2019-11-23 by todbot

SamboVisal, 2019-11-28:
Hi @todbot . `udevadm monitor` just display only 

`monitor will print the received events for:`
`UDEV - the event which udev sends out after rule processing`
`KERNEL - the kernel uevent
`

below is the my blink1.rules file which I followed all of your instruction

SUBSYSTEM=="input", GROUP="input", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="525", ATTRS{idProduct}=="a4ac", MODE:="666", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="525", ATTRS{idProduct}=="a4ac", MODE="0666", GROUP="plugdev"

I copied to /etc/udev/rules.d and then I unplugged the device (QR Scanner Device) and then run udevadm control --reload-rules && udevadm trigger as root. after that, I plug back the device and run `udevadm monitor`

todbot, 2019-11-28:
Hi @SamboVisal,
I'm unclear on your comment.  Are you still having a problem?

If so, is "525" the hex value of your VID?  If so, I think udev wants it to be formatted as "0525".  Or if "525" is the decimal value of VID (0x020d hex), then the udev idVendor property should be "020d".

If that doesn't work, try only having the "hidraw" line.   (and re-doing `udevadmn control --reload-rules`)

Finally, what does `sudo dmesg` show when you plug in the device?  It should show a line about a hidraw device being assigned.

SamboVisal, 2019-11-28:
Hello @todbot, Thank for your valuable reply.
Actually the vendor id is 0x525 and product id is 0xa4ac. I should convert them to hex and then place them in my rules file right?

The result of vendorId is 0525 and productId is a4ac. I am not sure whether the conversion is correct or not. 

And then I put those values in rules file 
SUBSYSTEM=="input", GROUP="input", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0525", ATTRS{idProduct}=="a4ac", MODE:="666", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="0525", ATTRS{idProduct}=="a4ac", MODE="0666", GROUP="plugdev".
After that, I unplugged the device and reload the rules file `sudo udevadmn control --reload-rules` and plug back the device and start `sudo npm start` and still display the message `Running as root without --no-sandbox is not supported.`
Could you please help check whether the values I put in idVendor and idProduct are correct or not? 
I also checked with your second solution by trying only the "hidraw" line as well but still not work.

todbot, 2019-11-28:
The whole point of setting up udev rules is so you do not need to run your program as the `root` user (e.g. `sudo ...`).  

In general you should not run `npm` and `node` as root as it can screw up the `npm` database.

This still sounds like a Node permissions problem, not a node-hid problem.