node-hid/node-hid #277: Make libusb debug logs configurable

webmaster128, 2018-08-24:
Heyho! Thanks for maintaining this lib!

When using node-hid on Linux, you always get those kind of messages written to STDOUT:

```
dev 413c / 2003 : usage_page:a8a8, usage:44c5
dev 2c97 / 1 : usage_page:2e31, usage:2f32
dev 46d / c24e : usage_page:2e34, usage:2f31
dev 46d / c24e : usage_page:2d32, usage:2e34
dev 413c / 2003 : usage_page:30, usage:0
dev 2c97 / 1 : usage_page:2e31, usage:2f32
dev 46d / c24e : usage_page:2e34, usage:2f31
dev 46d / c24e : usage_page:2d32, usage:2e34
dev 413c / 2003 : usage_page:30, usage:0
dev 2c97 / 1 : usage_page:2e31, usage:2f32
dev 46d / c24e : usage_page:2e34, usage:2f31
dev 46d / c24e : usage_page:2e34, usage:2f32
dev 413c / 2003 : usage_page:30, usage:0
dev 2c97 / 1 : usage_page:2e31, usage:2f32
dev 46d / c24e : usage_page:2e34, usage:2f31
dev 46d / c24e : usage_page:2e34, usage:2f32
```

Those debug messages come from libusb. libusb is statically linked into node-hid and node-hid provides prebuilds for my system. This meany my own system's libusb is not used. The libusb comes from ["testing VMs"](https://github.com/node-hid/node-hid/commit/00140f21556b234780cf9b86faa6da5e94acee22). If this libusb was [compiled with --enable-debug-log](https://stackoverflow.com/a/10250209), then the user has no chance to deactivate it.

A workaround for developers is to run

```
(cd node_modules/node-hid && node-gyp rebuild)
```

after the package installation to utilize their system's libusb.

You may want to have dynamically linked libusb to utilize the system's defaults and get updates for free without updating node-hid. But there are probably drawbacks with this approach.

This leads to the more general question about transparency of the build process. It would be good to know how those prebuilt binaries are created, on which machine, which OS, configuration.

todbot, 2019-02-14:
I don't understand this bug as `node-hid` uses hidraw, not libusb, on Linux.

As for how the prebuilds are created, they are created via the standard `prebuild` process as seen in `package.json` (e.g. `prebuild --all --verbose`).  This command is run on a Ubuntu 16 Linux box, a Windows 7 x64 box, and a Mac OS X box, all kept patched as current as possible.

webmaster128, 2019-02-15:
Interesting, thanks! Looks like we need a minimal example here as I only observed the behavior in an application that uses node-hid indirectly. I'll try to come up with one next time I work on the USB connectivity but cannot guarantee for any timeframe.

ethanfrey, 2019-02-15:
This PR seems like a work-around for the time being...

https://github.com/UltimateHackingKeyboard/agent/pull/853/files

closed 2019-11-23 by todbot