node-hid/node-hid #278: Electron

Wolrab1, 2018-09-04:
When adding node-hid to my electron app it runs fine in development but fails on production build.

When removing the package and running the build it works fine, when adding back the package it fails.

**My package.json** 

```
{
  "name": "electron-printer",
  "version": "0.1.0",
  "private": true,
  "scripts": {
    "serve": "vue-cli-service serve",
    "build": "vue-cli-service build",
    "build:electron": "vue-cli-service build:electron",
    "serve:electron": "vue-cli-service serve:electron",
    "postinstall": "electron-builder install-app-deps"
  },
  "dependencies": {
    "node-hid": "^0.7.3",
    "vue": "^2.5.17",
    "vue-barcode": "^1.1.0",
    "vue-router": "^3.0.1",
    "vuex": "^3.0.1"
  },
  "devDependencies": {
    "@vue/cli-plugin-babel": "^3.0.0",
    "@vue/cli-service": "^3.0.0",
    "electron": "^2.0.2",
    "electron-rebuild": "^1.8.2",
    "vue-cli-plugin-electron-builder": "^1.0.0-rc.2",
    "vue-template-compiler": "^2.5.17"
  },
  "main": "dist_electron/bundled/background.js"
}

```

**Error from console which doesn't give too much info,**

```
Unhandled rejection Error: Unresolved node modules: node-hid
    at F:\electron\PrinterTest\node_modules\app-builder-lib\src\util\packageDependencies.ts:108:17
From previous event:
    at Collector.resolveUnresolvedHoisted (F:\electron\PrinterTest\node_modules\app-builder-lib\src\util\packageDependencies.ts:164:10)
    at F:\electron\PrinterTest\node_modules\app-builder-lib\src\util\packageDependencies.ts:82:28
    at Generator.next (<anonymous>)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
From previous event:
    at Collector.collect (F:\electron\PrinterTest\node_modules\app-builder-lib\src\util\packageDependencies.ts:84:6)
    at F:\electron\PrinterTest\node_modules\app-builder-lib\out\util\packageDependencies.js:123:47
    at Generator.next (<anonymous>)
From previous event:
    at getProductionDependencies (F:\electron\PrinterTest\node_modules\app-builder-lib\out\util\packageDependencies.js:128:17)
    at Lazy.<anonymous> (F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:112:24)
    at Generator.next (<anonymous>)
From previous event:
    at Lazy.get value [as value] (F:\electron\PrinterTest\node_modules\lazy-val\src\main.ts:18:23)
    at F:\electron\PrinterTest\node_modules\app-builder-lib\src\util\yarn.ts:128:81
    at Generator.next (<anonymous>)
From previous event:
    at rebuild (F:\electron\PrinterTest\node_modules\app-builder-lib\out\util\yarn.js:239:18)
    at F:\electron\PrinterTest\node_modules\app-builder-lib\src\util\yarn.ts:20:11
    at Generator.next (<anonymous>)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
From previous event:
    at installOrRebuild (F:\electron\PrinterTest\node_modules\app-builder-lib\out\util\yarn.js:68:17)
    at F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:457:13
    at Generator.next (<anonymous>)
From previous event:
    at Packager.installAppDependencies (F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:418:70)
    at F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:368:20
    at Generator.next (<anonymous>)
From previous event:
    at Packager.doBuild (F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:344:39)
    at F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:314:57
    at Generator.next (<anonymous>)
    at F:\electron\PrinterTest\node_modules\graceful-fs\graceful-fs.js:99:16
    at F:\electron\PrinterTest\node_modules\graceful-fs\graceful-fs.js:43:10
    at FSReqWrap.oncomplete (fs.js:135:15)
From previous event:
    at Packager._build (F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:285:133)
    at F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:281:23
    at Generator.next (<anonymous>)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
From previous event:
    at Packager.build (F:\electron\PrinterTest\node_modules\app-builder-lib\src\packager.ts:238:14)
    at build (F:\electron\PrinterTest\node_modules\app-builder-lib\src\index.ts:58:28)
    at Object.build (F:\electron\PrinterTest\node_modules\electron-builder\src\builder.ts:227:10)
    at bundle.run (F:\electron\PrinterTest\node_modules\vue-cli-plugin-electron-builder\index.js:160:12)
    at finalCallback (F:\electron\PrinterTest\node_modules\webpack\lib\Compiler.js:210:39)
    at hooks.done.callAsync.err (F:\electron\PrinterTest\node_modules\webpack\lib\Compiler.js:259:14)
    at AsyncSeriesHook.eval [as callAsync] (eval at create (F:\electron\PrinterTest\node_modules\tapable\lib\HookCodeFactory.js:24:12), <anonymous>:6:1)
    at AsyncSeriesHook.lazyCompileHook [as _callAsync] (F:\electron\PrinterTest\node_modules\tapable\lib\Hook.js:35:21)
    at emitRecords.err (F:\electron\PrinterTest\node_modules\webpack\lib\Compiler.js:257:22)
    at Compiler.emitRecords (F:\electron\PrinterTest\node_modules\webpack\lib\Compiler.js:372:39)
    at emitAssets.err (F:\electron\PrinterTest\node_modules\webpack\lib\Compiler.js:251:10)
    at hooks.afterEmit.callAsync.err (F:\electron\PrinterTest\node_modules\webpack\lib\Compiler.js:358:14)
    at AsyncSeriesHook.eval [as callAsync] (eval at create (F:\electron\PrinterTest\node_modules\tapable\lib\HookCodeFactory.js:24:12), <anonymous>:6:1)
    at AsyncSeriesHook.lazyCompileHook [as _callAsync] (F:\electron\PrinterTest\node_modules\tapable\lib\Hook.js:35:21)
    at asyncLib.forEach.err (F:\electron\PrinterTest\node_modules\webpack\lib\Compiler.js:355:27)
    at done (F:\electron\PrinterTest\node_modules\neo-async\async.js:2854:11)
    at F:\electron\PrinterTest\node_modules\neo-async\async.js:2805:7
    at F:\electron\PrinterTest\node_modules\graceful-fs\graceful-fs.js:43:10
    at FSReqWrap.oncomplete (fs.js:135:15)
```

todbot, 2018-09-04:
That sounds like whatever packager you're using (`vue-cli-service`?) doesn't know or explicitly excludes native module library files.  

In webpack, I add a line like this:
```
  externals: {
    "node-hid": 'commonjs node-hid',
```
to enable node-hid to work on build products.

Wolrab1, 2018-09-05:
I'm using electron-builder, I've included my package.json. I'll put in a ticket with electron-builder and report back on any findings.

```
{
  "name": "electron-printer",
  "version": "0.1.0",
  "private": true,
  "scripts": {
    "serve": "vue-cli-service serve",
    "build": "vue-cli-service build",
    "build:electron": "vue-cli-service build:electron",
    "serve:electron": "vue-cli-service serve:electron",
    "postinstall": "electron-builder install-app-deps"
  },
  "dependencies": {
    "node-hid": "^0.7.3",
    "vue": "^2.5.17",
    "vue-barcode": "^1.1.0",
    "vue-router": "^3.0.1",
    "vuex": "^3.0.1"
  },
  "devDependencies": {
    "@vue/cli-plugin-babel": "^3.0.0",
    "@vue/cli-service": "^3.0.0",
    "electron": "^2.0.2",
    "electron-rebuild": "^1.8.2",
    "vue-cli-plugin-electron-builder": "^1.0.0-rc.2",
    "vue-template-compiler": "^2.5.17"
  },
  "main": "dist_electron/bundled/background.js"
}
 
```

john-yick, 2019-03-25:
@Wolrab1 Hi! Did you manage to resolve this issue? I am seeing the exact same issue using vue + electron + electron-builder + node-hid. Development works fine, but production builds do not work. I have tried as suggest adding

` 'node-hid': 'commonjs node-hid'`

[image]

todbot, 2019-06-19:
node-hid is know to work well in Electron. The above problems sound like module packing issues.  Refer to your packer's docs about how to deal with native modules.  

For working examples, see https://github.com/todbot/electron-hid-toy or https://github.com/todbot/Blink1Control2

closed 2019-06-19 by todbot

todbot, 2019-09-17:
Just found this about Vue and Electron:
https://nklayman.github.io/vue-cli-plugin-electron-builder/guide/guide.html#native-modules