node-hid/node-hid #281: Occasionally HID device not showing in HID.devices

ovsokolov, 2018-10-12:
We are using node-hid together with node-usb-detection with electron project.
Usually everything works fine but, but occasionally node-usb-detection detect device arrival but its not listed in HID.devices. What could be the reason for it?

todbot, 2018-10-13:
Hi,
Does the device eventually show up in `HID.devices()` or are you checking only once?
Depending on the OS, there could be a delay between when the device is recognized as being inserted and when it is recognized as a HID device.  Try inserting a delay of a hundred milliseconds or so before doing `HID.devices()`.

ovsokolov, 2018-10-13:
Yes its eventually appears in HID.devices(). Some times it takes long time, hundred milliseconds is not enough on some devices

closed 2018-10-13 by ovsokolov

DeekshithShetty, 2018-10-24:
@ovsokolov I am also facing this issue. May I know how you solved this?

ovsokolov, 2018-10-24:
I'm just keep waiting for device to appear on the list