node-hid/node-hid #282: Electron 3.0.X

ChaosRambo, 2018-10-18:
Hey,
if I update Electron to Electron 3.0.X and try to rebuild the app, I get an error message.
The rebuild works fine up to Electron 2.0.11.

You'll find an example app in my repository: [Busylight-Electron](https://github.com/ChaosRambo/Busylight-Electron)

I work with Ubuntu 18.04

Here is my Errorlog:

> Error: /usr/bin/node exited with code 1
> Output:
> node-hid@0.5.7 install /home/.../Busylight-Electron/node_modules/busylight/node_modules/node-hid
> prebuild-install || node-gyp rebuild

> make: Entering directory '/home/.../Busylight-Electron/node_modules/busylight/node_modules/node-hid/build'
>   CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
>   AR(target) Release/obj.target/hidapi.a
>   COPY Release/hidapi.a
>   CXX(target) Release/obj.target/HID/src/HID.o
> HID.target.mk:112: recipe for target 'Release/obj.target/HID/src/HID.o' failed
> make: Leaving directory '/home/.../Busylight-Electron/node_modules/busylight/node_modules/node-hid/build'

> Error output:
> prebuild-install WARN install No prebuilt binaries found (target=3.0.4 runtime=electron arch=x64 platform=linux)
> ../hidapi/libusb/hid.c: In function ‘hid_read_timeout’:
> ../hidapi/libusb/hid.c:1078:6: warning: variable ‘bytes_read’ might be clobbered by ‘longjmp’ or ‘vfork’ [-Wclobbered]
  int bytes_read = -1;
      ^~~~~~~~~~
../src/HID.cc:67:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     throw(JSException);
     ^~~~~
../src/HID.cc:70:5: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
     throw(JSException);
     ^~~~~
../src/HID.cc:152:3: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
   throw(JSException)
   ^~~~~
../src/HID.cc:163:3: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
   throw(JSException)
   ^~~~~
../src/HID.cc: In member function ‘void HID::readResultsToJSCallbackArguments(HID::ReceiveIOCB*, v8::Local<v8::Value>*)’:
../src/HID.cc:207:74: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
     Local<Object> buf = nodeBufConstructor->NewInstance(1, nodeBufferArgs);
                                                                          ^
In file included from /home/.../.electron-gyp/iojs-3.0.4/src/node.h:54:0,
                 from ../src/HID.cc:30:
/home/.../.electron-gyp/iojs-3.0.4/deps/v8/include/v8.h:3853:44: note: candidate: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>, int, v8::Local<v8::Value>*) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                            ^~~~~~~~~~~
/home/.../.electron-gyp/iojs-3.0.4/deps/v8/include/v8.h:3853:44: note:   candidate expects 3 arguments, 2 provided
/home/.../.electron-gyp/iojs-3.0.4/deps/v8/include/v8.h:3856:44: note: candidate: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                            ^~~~~~~~~~~
/home/.../.electron-gyp/iojs-3.0.4/deps/v8/include/v8.h:3856:44: note:   candidate expects 1 argument, 2 provided
../src/HID.cc: In static member function ‘static void HID::recvAsyncDone(uv_work_t*)’:
../src/HID.cc:231:32: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated [-Wdeprecated-declarations]
   iocb->_callback->Call(2, argv);
                                ^
In file included from ../src/HID.cc:31:0:
../../../../nan/nan.h:1655:3: note: declared here
   Call(int argc, v8::Local<v8::Value> argv[]) const {
   ^~~~
../src/HID.cc: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE HID::New(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/HID.cc:409:62: warning: ‘v8::String::Value::Value(v8::Local<v8::Value>)’ is deprecated: Use Isolate version [-Wdeprecated-declarations]
         serialPointer = (wchar_t*) *v8::String::Value(info[2]);
                                                              ^
In file included from /home/.../.electron-gyp/iojs-3.0.4/deps/v8/include/v8.h:26:0,
                 from /home/.../.electron-gyp/iojs-3.0.4/src/node.h:54,
                 from ../src/HID.cc:30:
/home/.../.electron-gyp/iojs-3.0.4/deps/v8/include/v8.h:2844:51: note: declared here
     V8_DEPRECATED("Use Isolate version", explicit Value(Local<v8::Value> obj));
                                                   ^
/home/.../.electron-gyp/iojs-3.0.4/deps/v8/include/v8config.h:318:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
make: *** [Release/obj.target/HID/src/HID.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.15.0-36-generic
gyp ERR! command "/usr/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/.../Busylight-Electron/node_modules/busylight/node_modules/node-hid
gyp ERR! node -v v8.10.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-hid@0.5.7 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/.../.npm/_logs/2018-10-18T08_07_22_370Z-debug.log

    at ChildProcess.childProcess.once.code (/home/.../Busylight-Electron/node_modules/builder-util/src/util.ts:254:14)
    at Object.onceWrapper (events.js:317:30)
    at emitTwo (events.js:126:13)
    at ChildProcess.emit (events.js:214:7)
    at maybeClose (internal/child_process.js:925:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
From previous event:
    at rebuild (/home/.../Busylight-Electron/node_modules/app-builder-lib/out/util/yarn.js:239:18)
    at /home/.../Busylight-Electron/node_modules/app-builder-lib/src/util/yarn.ts:20:11
From previous event:
    at installOrRebuild (/home/.../Busylight-Electron/node_modules/app-builder-lib/out/util/yarn.js:68:17)
    at /home/.../Busylight-Electron/node_modules/electron-builder/src/cli/install-app-deps.ts:56:9
    at Generator.next (<anonymous>)
From previous event:
    at installAppDeps (/home/.../Busylight-Electron/node_modules/electron-builder/out/cli/install-app-deps.js:174:17)
    at then (/home/.../Busylight-Electron/node_modules/electron-builder/src/cli/cli.ts:42:48)
    at runCallback (timers.js:794:20)
    at tryOnImmediate (timers.js:752:5)
    at processImmediate [as _immediateCallback] (timers.js:729:5)
From previous event:
    at Object.args [as handler] (/home/.../Busylight-Electron/node_modules/electron-builder/src/cli/cli.ts:42:48)
    at Object.runCommand (/home/.../Busylight-Electron/node_modules/yargs/lib/command.js:238:44)
    at Object.parseArgs [as _parseArgs] (/home/.../Busylight-Electron/node_modules/yargs/yargs.js:1059:30)
    at Object.get [as argv] (/home/.../Busylight-Electron/node_modules/yargs/yargs.js:1000:21)
    at Object.<anonymous> (/home/.../Busylight-Electron/node_modules/electron-builder/src/cli/cli.ts:25:28)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:188:16)
    at bootstrap_node.js:609:3

ChaosRambo, 2018-10-18:
Thanks for your help

Best regards
ChaosRambo

hamaluik, 2018-11-02:
Same issue here

webees, 2018-12-18:
me too too too too too too too……

todbot, 2018-12-18:
From what I can tell, this is a bug in the particular version of Node v10 used in Electron 3.
See this issue: https://github.com/nodejs/nan/issues/832

todbot, 2018-12-18:
And it looks like a work-around to try is to add this to your top-level `package.json` dependencies:
```
  "nan": "github:bnoordhuis/nan#fix832",
```

setpixel, 2018-12-29:
@todbot It also doesn't work in Electron 4.0.0 (the latest main release). I have an issue here: https://github.com/node-hid/node-hid/issues/293

todbot, 2019-02-14:
this is basically dup of #293

closed 2019-02-14 by todbot