node-hid/node-hid #283: How to make a Symbol/Motorola/Zebra Barcodescanner beep?

luxlux, 2018-10-18:
I use hode-hid to read scans from some Symbol/Motorola/Zebra DS-6708 barcode scanners.
But Node-hid offers also the possibility to write to a HID device.
Does any body know what exactly i have to write to such an scanner, to let the scanner beep once, twice, ...?

Or where I could find how to do this?

todbot, 2018-10-18:
If you can get the protocol specs for your scanner, it may be possible to figure out what commands to sent to it.  But without that, it's not possible.

luxlux, 2018-10-19:
I don't have protocal spects. Zebra only offers an SDK which works if you install a middleware, which i don't want to use. So i guess the spects are only implemented in the middleware. Anyhow i have used the example app from the SDK (with the middleware installed) which also allows to let the scanner beep and used wireshark to capture the USB communication. I send different beep commands from the GUI to the scanner. In the capture only one byte differs. but unfortunately i can't interpret the rest of the packet. i don't know where the message starts and what is the overhead. and i also don't know, if they simply write the the device or do something with FeatureReports, or what ever.

anyhow, if somebody knows about this about ZEBRA at a later time, please simply answer here.

lafar6502, 2019-09-20:
if you have serial port or serial port emulation on COM and the scanner is connected in serial mode then you can send ASCII BEL character over the serial port to trigger a beep.

luxlux, 2019-09-20:
@lafar6502 Thanks for your message.

Yes I know, BEL  (HEX: 07) should work on a serial connection. 
But how can I send such a message via HID-Node in the right way?

If I try a `console.log(scanner.write(0x07))` electron will be shut down without a error message.
If I try a `console.log(scanner.write([0x07]))` electron throws an Uncaught Exception: "Can not write to HID device".

BTW: receiving data with `scanner.on("data", function(data) { ...})` works.

luxlux, 2019-09-20:
I remember, may be the reason is I use the scanner set into "Simple COM Port Emulation" mode, but not in "USB Keyboard HID" mode. Anyhow, reading works this way.

Unfortunately i can not use the "USB Keyboard HID" mode. If i set the scanner into this mode, it will be consumed by Mac OS and is not accessible by NODE-HID...

todbot, 2019-09-20:
If you're in serial port mode, you can use `node-serialport` to talk to the device.

closed 2019-11-23 by todbot