node-hid/node-hid #284: Can I get the IP of a HID device?

piavgh, 2018-10-23:
Hi,

Can you please tell me how can I get the IP of HID device using this library?

Thank you very much.

todbot, 2018-10-26:
Hi,
I'm not sure what you mean. USB devices do not have IP addresses.

piavgh, 2018-10-28:
@todbot : Hi, I'm working on a project for a client. That's what I thought also when looking at the requirement

But this is the context that my client provided:

"The device (1) is a touch display with a built-in PC
the display itself has an Android operating system

On this Android system, there is an application called AppOverlay
AppOverlay can send out data, it uses WebSocket for this

Notes+ (another app) needs to make a connection to this WebSocket
by default, the IP of the Android device is 192.168.55.1
Notes+ will continuously try to connect to that IP address
so far so good, for a "default" system
but the default IP address may cause issues in some network environments
therefore the IP address of the device (1) can be changed
in that case, Notes+ can no longer connect

To solve this, Notes+ should obtain the new IP address and use that instead
And the IP address can be obtained by the "MCU", this is an HID device connected to the PC"

Can you understand this context? If even you can't,  I think that I will need to reject this task from my client

Thanks

todbot, 2019-06-19:
Sorry can't help you.  Hope you find out what the client is talking about.

closed 2019-06-19 by todbot