node-hid/node-hid #285: Cannot connect to device which gets plugged after app start

coldner, 2018-10-29:
When I start app with plugged device I can connect, but when I plug any device after start I see it in HID.devices() but I cannot connect.

todbot, 2018-10-29:
Hi,
Please provide more details. Such as:
- what these devices are
- what OS you're on
- a code snippet showing what you're trying to do
- the exact error message you are receiving
Thanks!

coldner, 2018-10-30:
I think it doesn't work because I run my app in docker. Docker doesn't see new USB device which plugs after container start... do you have any ideas?

todbot, 2018-10-30:
I think you will need to start the VirtualBox Manager on the docker VM and choose to add USB support.  I've not tried this.

todbot, 2019-06-19:
Any update? Closing for now. Let me know if you have any other questions.

closed 2019-06-19 by todbot