node-hid/node-hid #292: arm64/android compilation issues

Aicos, 2018-11-20:
I'm having problems getting this to build on my arm64/android with errors like this:

```
/Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC
```

This is cross compiled from Cordova 7. (also scroll to the bottom of the post for more info)

Here’s the build log:

```
$ cordova build android

ANDROID_HOME=/Users/aicos/android/
JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.8.0_192.jdk/Contents/Home
Subproject Path: CordovaLib
Starting a Gradle Daemon (subsequent builds will be faster)
/Users/aicos/.nvm/versions/node/v8.2.1/bin/npm
The Task.leftShift(Closure) method has been deprecated and is scheduled to be removed in Gradle 5.0. Please use Task.doLast(Action) instead.
	at build_dgnimap2xmuwmli9y3r3di9to.run(/Users/aicos/proj/uploader2/platforms/android/build.gradle:142)
The JavaCompile.setDependencyCacheDir() method has been deprecated and is scheduled to be removed in Gradle 4.0.
Incremental java compilation is an incubating feature.
The TaskInputs.source(Object) method has been deprecated and is scheduled to be removed in Gradle 4.0. Please use TaskInputs.file(Object).skipWhenEmpty() instead.
:CopyNodeProjectAssetsarmeabi-v7a
:MakeToolchainarmeabi-v7a UP-TO-DATE
:BuildNpmModulesarmeabi-v7anpm info it worked if it ends with ok
npm verb cli [ '/Users/aicos/.nvm/versions/node/v8.2.1/bin/node',
npm verb cli   '/Users/aicos/.nvm/versions/node/v8.2.1/bin/npm',
npm verb cli   '--verbose',
npm verb cli   'rebuild',
npm verb cli   '--build-from-source' ]
npm info using npm@5.3.0
npm info using node@v8.2.1
npm info readInstalled object
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project',
npm verb rebuild   'nodejs-mobile-sample-project@0.1.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/nan',
npm verb rebuild   'nan@2.9.2' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-hid',
npm verb rebuild   'node-hid@0.7.3' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/bindings',
npm verb rebuild   'bindings@1.3.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-hid/node_modules/nan',
npm verb rebuild   'nan@2.11.1' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-hid/node_modules/prebuild-install',
npm verb rebuild   'prebuild-install@4.0.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/detect-libc',
npm verb rebuild   'detect-libc@1.0.3' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/expand-template',
npm verb rebuild   'expand-template@1.1.1' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/github-from-package',
npm verb rebuild   'github-from-package@0.0.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/minimist',
npm verb rebuild   'minimist@1.2.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/mkdirp',
npm verb rebuild   'mkdirp@0.5.1' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/mkdirp/node_modules/minimist',
npm verb rebuild   'minimist@0.0.8' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-abi',
npm verb rebuild   'node-abi@2.5.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/semver',
npm verb rebuild   'semver@5.6.0' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/noop-logger',
npm verb rebuild   'noop-logger@0.1.1' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/npmlog',
npm verb rebuild   'npmlog@4.1.2' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/are-we-there-yet',
npm verb rebuild   'are-we-there-yet@1.1.5' ]
npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/delegates',
npm verb rebuild   'delegates@1.0.0' ]
[... 465 lines cut ...]
/Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC
/Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC
/Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC
/Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC
/Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC
/Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC
/Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC
clang70++: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Release/obj.target/HID.node] Error 1

```

Recompiling with -fPIC got rid of the errors.  But  running the app resulted in a crash with the following error:

```
11-20 11:06:27.408  7483  7584 E NODEJS-MOBILE: Error: dlopen failed: cannot locate symbol "hid_open" referenced by "/data/data/com.example.hello/files/www/nodejs-project/node_modules/node-hid/build/Release/HID.node"...
```

One final point; I've tried compiling on Termex (aarch64) from the device itself and I'm getting errors like this:

```
/data/data/com.termux/files/usr/bin/aarch64-linux-android-ld: Release/obj.target/HID/src/HID.o: relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `_ZTCNSt6__ndk119basic_ostringstreamIcNS_11char_traitsIcEENS_9allocatorIcEEEE0_NS_13basic_ostreamIcS2_EE' which may bind externally can not be used when making a shared object; recompile with -fPIC
```

Thanks in advance for any responses.

todbot, 2019-02-16:
Can you try `npm install node-hid --build-from-source -driver=libusb`?

todbot, 2019-07-25:
Closed because of inactivity

closed 2019-07-25 by todbot

diptasarkarr, 2021-12-10:
build_dgnimap2xmuwmli9y3r3di9to.run(/Users/aicos/proj/uploader2/platforms/android/build.gradle:142) The JavaCompile.setDependencyCacheDir() method has been deprecated and is scheduled to be removed in Gradle 4.0. Incremental java compilation is an incubating feature. The TaskInputs.source(Object) method has been deprecated and is scheduled to be removed in Gradle 4.0. Please use TaskInputs.file(Object).skipWhenEmpty() instead. :CopyNodeProjectAssetsarmeabi-v7a :MakeToolchainarmeabi-v7a UP-TO-DATE :BuildNpmModulesarmeabi-v7anpm info it worked if it ends with ok npm verb cli [ '/Users/aicos/.nvm/versions/node/v8.2.1/bin/node', npm verb cli '/Users/aicos/.nvm/versions/node/v8.2.1/bin/npm', npm verb cli '--verbose', npm verb cli 'rebuild', npm verb cli '--build-from-source' ] npm info using npm@5.3.0 npm info using node@v8.2.1 npm info readInstalled object npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project', npm verb rebuild 'nodejs-mobile-sample-project@0.1.0' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/nan', npm verb rebuild 'nan@2.9.2' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-hid', npm verb rebuild 'node-hid@0.7.3' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/bindings', npm verb rebuild 'bindings@1.3.0' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-hid/node_modules/nan', npm verb rebuild 'nan@2.11.1' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-hid/node_modules/prebuild-install', npm verb rebuild 'prebuild-install@4.0.0' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/detect-libc', npm verb rebuild 'detect-libc@1.0.3' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/expand-template', npm verb rebuild 'expand-template@1.1.1' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/github-from-package', npm verb rebuild 'github-from-package@0.0.0' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/minimist', npm verb rebuild 'minimist@1.2.0' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/mkdirp', npm verb rebuild 'mkdirp@0.5.1' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/mkdirp/node_modules/minimist', npm verb rebuild 'minimist@0.0.8' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/node-abi', npm verb rebuild 'node-abi@2.5.0' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/semver', npm verb rebuild 'semver@5.6.0' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/noop-logger', npm verb rebuild 'noop-logger@0.1.1' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/npmlog', npm verb rebuild 'npmlog@4.1.2' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/are-we-there-yet', npm verb rebuild 'are-we-there-yet@1.1.5' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/delegates', npm verb rebuild 'delegates@1.0.0' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/readable-stream', npm verb rebuild 'readable-stream@2.3.6' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build/nodejs-native-assets-temp-build/nodejs-native-assets-armeabi-v7a/nodejs-project/node_modules/core-util-is', npm verb rebuild 'core-util-is@1.0.2' ] npm verb rebuild path, id [ '/Users/aicos/proj/uploader2/platforms/android/build [... cut ...] c/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC /Users/aicos/proj/uploader2/platforms/android/build/standalone-toolchains/arm-linux-androideabi/bin/../lib/gcc/arm-linux-androideabi/4.9.x/../../../../arm-linux-androideabi/bin/ld: error: Release/obj.target/HID/src/HID.o: requires unsupported dynamic reloc R_ARM_REL32; recompile with -fPIC clang70++: error: linker command failed with exit code 1 (use -v to see invocation) make: *** [Release/obj.target/HID.node] Error 1