node-hid/node-hid #293: node-hid does not work with latest Electron release.

setpixel, 2018-12-29:
Electron 4.0.0 was just released. It is the main release.

Electron 4.0.0 uses node 10.11.0.

It seems to build fine with rebuild, and it works when I require.

However, when I try to call hid.devices(), i get:

`
dyld: lazy symbol binding failed: Symbol not found: __ZN2v811HandleScope12CreateHandleEPNS_8internal10HeapObjectEPNS1_6ObjectE
  Referenced from: /Users/setpixel/git/shot-generator/node_modules/node-hid/build/Release/HID.node
  Expected in: flat namespace

dyld: Symbol not found: __ZN2v811HandleScope12CreateHandleEPNS_8internal10HeapObjectEPNS1_6ObjectE
  Referenced from: /Users/setpixel/git/shot-generator/node_modules/node-hid/build/Release/HID.node
  Expected in: flat namespace
`

todbot, 2018-12-29:
Hi,
Which OS are you on?

todbot, 2018-12-29:
Verified this on MacOS.  Closed issue by accident.

setpixel, 2018-12-29:
Yes mac, sorry I thought it was implied by the path, but I guess linux could be the same. sorry!

setpixel, 2018-12-29:
btw, thanks for your work on this module, it's great!!!!

todbot, 2018-12-29:
Thanks! It's unfortunate it's such a pain that it seems to always break whenever a new Node or Electron version comes out. Seems the way for native Node modules though.

todbot, 2018-12-29:
Note to self: I think this error message is essentially a Node version mismatch error between command-line Node v10 and the Node v10 inside of Electron.  

That is, it seems like this is a problem with `electron-rebuild` (which hasn't had a new release since 30 Jul 2018)

todbot, 2018-12-29:
I think I have a work-around fix for this.
You can check it out in [todbot/electron-hid-toy](https://github.com/todbot/electron-hid-toy)
You should be able to do:
```
git clone https://github.com/todbot/electron-hid-toy
cd electron-hid-toy
npm install
npm run start
```
and see a list of HID devices, running in an Electron v4.0.0 window.

todbot, 2018-12-29:
Note to self: It seems the main issue is that node-hid@0.7.4 has prebuilt binaries for Node v10 and Electron v3.x, but Node v10 is such a mess wrt native modules that `electron-rebuild` was getting confused, loading the prebuild for 0.7.4 for Electron v4 and not building from source.  I think.  Still investigating.

todbot, 2018-12-30:
`node-hid@0.7.6` released but prebuilds for Electron v4.0.0 still do not work correctly and give that  `dyld: lazy symbol binding failed`.  This seems to be a failure of `prebuild`.
Current strategy: remove prebuilds for Electron v4 to enforce node-hid to be recompiled on `npm install`.  This means you need a C compiler dev environment. See: https://github.com/node-hid/node-hid#to-build-node-hid-for-development

gniezen, 2019-01-09:
There seems to be more details on this issue here: https://github.com/lgeiger/node-abi/pull/55

TL;DR: The ABI for both Electron v3 and v4 is 64, but the builds are not compatible (probably because they switched from GYP to GN).

todbot, 2019-01-09:
Thanks for finding that @gniezen! I was going down similar rabbit holes but gave up.

adeperio, 2019-02-14:
Hi @todbot the implementation in todbot/electron-hid-toy works well for me with my mac builds (packaging and executing). I can also generate a windows build (which I am building on mac), but when I run the .exe that's when I run into problems. 

It doesn't run on windows and I keep getting an error message on a file in the temp folder which says "this is not a valid win32 application"

Could this be something todo with bypassing using prebuilds?

[image]

todbot, 2019-02-14:
Hi @adeperio, what version of Windows are you running?

adeperio, 2019-02-14:
Hi @todbot, I'm running Win7 64bit and building specifically for that with electron-builder. I also have another native dependency (keytar) so I'm not sure if node-hid is the problem or if its keytar

todbot, 2019-02-14:
Oh wait you are building a Windows executable on Mac?  In general that seems to never work well.  Try building on Windows

adeperio, 2019-02-14:
@todbot Yea I've been building windows on Mac fro a while now. I've only started having problems after upgrading to Electron 4. I'd like to continue building Windows on Mac if possible. I'm not sure where to look for any changes in the compatibility between the upgrade to Electron 4 (from 1.8.6, where keytar was working with windows being built on mac) to the latest keytar. Is there anything else you think I could look at?

adeperio, 2019-02-14:
@todbot I think I was using the prebuilt binaries before with keytar, and the build seemed to be working with those. I noticed that prebuilt binaries are not included with keytar 4.4.0? If that's the case, any idea if they may be coming out any time soon? Or if not, is there a way I can generate those prebuilt binaries myself?

todbot, 2019-02-14:
yeah because of the `node-abi` bug referenced above, it's difficult currently to make Electron prebuilds for Electron 3 & 4.  And that means we can't do Windows builds of projects with native modules on Mac (because you'd need a Windows compiler on Mac)

(although now that I look at that bug, perhaps the problem has been fixed as of a few days ago with Electron 4.0.4 that incorporates this fix: https://github.com/electron/electron/pull/16687 )

I'll do some tests tomorrow.

adeperio, 2019-02-15:
@todbot Thanks Tod, I've been following the comments around this ABI issue. Hopefully 4.0.4 has a fix? It would be useful to get new prebuilds with node-hid. If it helps, I was packaging windows up successfully on a mac machine prior to this on Electron 1.8.6 using the node-hid prebuilds. But we're now looking to upgrade from 1.8.6 to 4.0.4 and hence started running into this issue

webees, 2019-02-15:
```
usb-detection: Command failed.
rebuild-install WARN install No prebuilt binaries found (target=4.0.4 runtime=electron arch=ia32 libc= platform=win32)

```

sys: win10
electron: v4.0.4
npm: [ 
 node-hid
 usb-detection
]

No problem with electron v2.0.15

todbot, 2019-02-15:
that looks like a problem with `usb-detection`, not `node-hid`. 
I've not published an updated version of `node-hid` yet, apologies.

adeperio, 2019-02-15:
Hi @todbot thank you, I actually just saw 0.7.7 come up on the releases page. Will look out for it on npm when it's available there and see if that brings me a step forward on my end

webees, 2019-02-15:
can't find version 0.7.7 in npm.

webees, 2019-02-15:
The version of usb-detection must be less than v3.0.0, It is used with node-hid. Is there a substitute for usb-detection?

webees, 2019-02-15:
'user-agent': 'yarn/1.13.0 node/v8.15.0 win32 ia32',

[image]

todbot, 2019-02-15:
I believe this is fixed now thanks to Electron v4.0.4+. 
Tested with an updated https://github.com/todbot/electron-hid-toy 
Be sure to update your `package.json` to include updated `electron`, `electron-rebuild`, and `electron-builder`.

closed 2019-02-15 by todbot

adeperio, 2019-02-16:
Thanks @todbot can confirm from my end that 0.7.7 node-hid is now working with Electron 4.0.4!

commit 39970452 (2018-12-29): update nan and bindings to potentially fix #293