node-hid/node-hid #294: Device serialNumber with non ASCII

ovsokolov, 2019-01-09:
Hi,

We have device with serialNumber which has non asci charachter.
For example one of our devices has last byte with 0x1D as last byte
When we read HD device list it shows
'UM1321\u001D'
So we can take last byte and using charCodeAt get value 27 
but if or last bite is 0x9B (155 dec) the HID  shows in console
'UM1321� '
it doesnt show it in unicode and we can not find anyway to get actual number out of it.
Does hid-node convert such number properly?
How we can get value out of it if it does

todbot, 2019-01-09:
node-hid uses standard Unicode UTF-16 to UTF-8 conversion routines to convert the UTF-16-formatted USB strings to format Node can use.  
I'm not quite sure why you're using 27 to even get the last char of that UTF-16 string. In UTF-16, I believe the character sequence "UM1321\u001D" is 14 bytes long, e.g.
`{ 0, 'U', 0, 'M', 0, '1', 0, '3', 0, '2', 0, '1', 0, 0x1D }` 
in C array format. 

In UTF-8, I believe that character array becomes 7 bytes long. So shouldn't you be doing `serial.charCodeAt(7)` to get the last byte of your serial number?

todbot, 2019-01-09:
I do not understand.  From your example serial number "UM1321\u001D", it is only seven characters long, with the last character being 0x1D.

todbot, 2019-01-09:
oh do you mean if you change 0x1D to be 0x9B?  I think that is not allowed in UTF-8.

ovsokolov, 2019-01-09:
Oh, ok that is working as expected in our another app which is C++. You are right 0x9B not valid UTF-8, we were hoping it would work the same

closed 2019-11-23 by todbot