node-hid/node-hid #296: Segfault on hid_write

ppershing, 2019-01-31:
Hi,
 while running some exhaustive tests which interact with Ledger via node-hid, I come across reproducible crash. I managed to get the following stack snapshot.

Node-hid version 0.7.6

```
/home/ppershing/vl/ledger/ledgerjs/node_modules/segfault-handler/build/Release/segfault-handler.node(+0x2fa6)[0x7fc56c870fa6]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x12670)[0x7fc56c519670]
/home/ppershing/vl/ledger/ledgerjs/node_modules/node-hid/build/Release/HID.node(hid_write+0x1)[0x7fc5690c2031]
/home/ppershing/vl/ledger/ledgerjs/node_modules/node-hid/build/Release/HID.node(_ZN3HID5writeERKSt6vectorIhSaIhEE+0x353)[0x7fc5690bd593]
/home/ppershing/vl/ledger/ledgerjs/node_modules/node-hid/build/Release/HID.node(_ZN3HID5writeERKN3Nan20FunctionCallbackInfoIN2v85ValueEEE+0x2a3)[0x7fc5690bd963]
/home/ppershing/vl/ledger/ledgerjs/node_modules/node-hid/build/Release/HID.node(+0x5ab5)[0x7fc5690baab5]
/home/ppershing/.nvm/versions/node/v8.11.1/bin/node(_ZN2v88internal25FunctionCallbackArguments4CallEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEE+0x193)[0xb03853]
/home/ppershing/.nvm/versions/node/v8.11.1/bin/node[0xb7a9ac]
/home/ppershing/.nvm/versions/node/v8.11.1/bin/node(_ZN2v88internal21Builtin_HandleApiCallEiPPNS0_6ObjectEPNS0_7IsolateE+0xaf)[0xb7b5ff]
[0x8d5108842fd]
```

Is there any way I can help with further debugging?

todbot, 2019-01-31:
Hi @ppershing, 
Yes I believe there is a race condition if you try to re-use a closed HID object.  I believe this is down inside of the `hidapi` library that we don't control.  You're seeing this on Mac, I assume?

ppershing, 2019-01-31:
Hi @todbot No, I am seeing this on Debian. As for the race condition -- isn't there an easy way to fix it on the javascript side (maybe do some additional bookkeeping)?

todbot, 2019-01-31:
As far as I can tell there's no good way. I looked into this a year ago or so (`hidapi` doesn't get updated very often unfortunately). There may be a small null check we could do (see issue https://github.com/node-hid/node-hid/issues/230) but I've yet to find a small self-contained example that exhibits the problem to test against.

todbot, 2019-11-23:
`node-hid` has been updated to [libusb/hidapi](https://github.com/libusb/hidapi), which includes many bug fixes.   I'm going to assume this is fixed unless I hear otherwise. :)

closed 2019-11-23 by todbot