node-hid/node-hid #297: Bypass Windows keyboard limitations

justablob, 2019-02-04:
Hey, so i was looking for a HID library to add custom lighting to my K70 LUX, but upon trying, only the media keys seemed to trigger an event. All other keys didn't emit an event. The keyboard also showed up 6 times in the HID list, but i tried to listen on all of them and not a single one worked. What i want to do is trigger a kind of "Ripple effect" around the key i press, but obviously i need to listen for any keystroke to do that. Then i pulled up this project and saw that this is actually a Windows limitation. Is there any way around that? Say, for example, emulating a second HID device and passing all keystrokes from the actual keyboard to it?

justablob, 2019-02-04:
I also thought of using usbpcap but i'd like to make sure there's no easier way before i do that.

todbot, 2019-02-14:
You would be breaking basic OS system security procedures because you're essentially talking about making a keylogger.  So, no there's no way for node-hid to do this.  

But as you say, you could potentially unload the kernel driver for a particular device, install a new HID driver of your own devising, and have a pcap-like program that listens/injects data into the communication between the keyboard and your custom HID driver.

closed 2019-02-14 by todbot