node-hid/node-hid #298: Error building node-hid

mohammedabbas27, 2019-02-06:
I am trying to install node-hid on raspberry pi and when im trying to compile node-hid using the following command : pi@raspberrypi:~/node-hid $ npm install --build-from-source
I get the following error:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'install' ]
2 info using npm@6.4.1
3 info using node@v8.15.0
4 verbose npm-session 41e1d74e2e518021
5 silly install runPreinstallTopLevelLifecycles
6 silly preinstall node-hid@0.7.6
7 info lifecycle node-hid@0.7.6~preinstall: node-hid@0.7.6
8 silly install loadCurrentTree
9 silly install readLocalPackageData
10 timing stage:loadCurrentTree Completed in 3089ms
11 silly install loadIdealTree
12 silly install cloneCurrentTreeToIdealTree
13 timing stage:loadIdealTree:cloneCurrentTree Completed in 30ms
14 silly install loadShrinkwrap
15 timing stage:loadIdealTree:loadShrinkwrap Completed in 1708ms
16 silly install loadAllDepsIntoIdealTree
17 timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 1786ms
18 timing stage:loadIdealTree Completed in 4197ms
19 silly currentTree node-hid@0.7.6
19 silly currentTree +-- abbrev@1.1.1
19 silly currentTree +-- after@0.8.2
19 silly currentTree +-- ajv@6.8.1
19 silly currentTree +-- amdefine@1.0.1
19 silly currentTree +-- ansi-regex@2.1.1
19 silly currentTree +-- ansi@0.3.1
19 silly currentTree +-- aproba@1.2.0
19 silly currentTree +-- are-we-there-yet@1.1.5
19 silly currentTree +-- array-index@1.0.0
19 silly currentTree +-- asn1@0.2.4
19 silly currentTree +-- assert-plus@1.0.0
19 silly currentTree +-- async@2.6.1
19 silly currentTree +-- asynckit@0.4.0
19 silly currentTree +-- aws-sign2@0.7.0
19 silly currentTree +-- aws4@1.8.0
19 silly currentTree +-- balanced-match@1.0.0
19 silly currentTree +-- bcrypt-pbkdf@1.0.2
19 silly currentTree +-- big-integer@1.6.41
19 silly currentTree +-- binary@0.3.0
19 silly currentTree +-- bindings@1.4.0
19 silly currentTree +-- bl@1.2.2
19 silly currentTree +-- block-stream@0.0.9
19 silly currentTree +-- bluebird@2.11.0
19 silly currentTree +-- brace-expansion@1.1.11
19 silly currentTree +-- buffer-alloc-unsafe@1.1.0
19 silly currentTree +-- buffer-alloc@1.2.0
19 silly currentTree +-- buffer-fill@1.0.0
19 silly currentTree +-- buffer-from@0.1.2
19 silly currentTree +-- buffer-indexof-polyfill@1.0.1
19 silly currentTree +-- buffer-shims@1.0.0
19 silly currentTree +-- buffers@0.1.1
19 silly currentTree +-- camelcase@2.1.1
19 silly currentTree +-- caseless@0.12.0
19 silly currentTree +-- chainsaw@0.1.0
19 silly currentTree +-- chownr@1.1.1
19 silly currentTree +-- cliui@3.2.0
19 silly currentTree +-- cmake-js@3.7.3
19 silly currentTree | +-- are-we-there-yet@1.0.6
19 silly currentTree | +-- gauge@1.2.7
19 silly currentTree | +-- lodash@3.10.1
19 silly currentTree | `-- npmlog@1.2.1
19 silly currentTree +-- code-point-at@1.1.0
19 silly currentTree +-- combined-stream@1.0.7
19 silly currentTree +-- commander@2.9.0
19 silly currentTree +-- concat-map@0.0.1
19 silly currentTree +-- console-control-strings@1.1.0
19 silly currentTree +-- core-util-is@1.0.2
[... 482 lines cut ...]
59 silly lifecycle node-hid@0.7.6~install: Returned: code: 1  signal: null
60 info lifecycle node-hid@0.7.6~install: Failed to exec install script
61 verbose stack Error: node-hid@0.7.6 install: `prebuild-install || node-gyp rebuild`
61 verbose stack Exit status 1
61 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
61 verbose stack     at emitTwo (events.js:126:13)
61 verbose stack     at EventEmitter.emit (events.js:214:7)
61 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
61 verbose stack     at emitTwo (events.js:126:13)
61 verbose stack     at ChildProcess.emit (events.js:214:7)
61 verbose stack     at maybeClose (internal/child_process.js:915:16)
61 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
62 verbose pkgid node-hid@0.7.6
63 verbose cwd /home/pi/node-hid
64 verbose Linux 4.14.79-v7+
65 verbose argv "/usr/bin/node" "/usr/bin/npm" "install"
66 verbose node v8.15.0
67 verbose npm  v6.4.1
68 error code ELIFECYCLE
69 error errno 1
70 error node-hid@0.7.6 install: `prebuild-install || node-gyp rebuild`
70 error Exit status 1
71 error Failed at the node-hid@0.7.6 install script.
71 error This is probably not a problem with npm. There is likely additional logging output above.
72 verbose exit [ 1, true ]

todbot, 2019-02-06:
Hi,
Which Raspberry Pi OS are you running?

It's hard to read your error because of the formatting (and I think it's not showing the actual error you may've seen in your terminal), but I think it's showing you don't have a C/C++ build environment (compiler,etc) installed.  

Try installing a build environment for node-hid with something like:
`sudo apt install build-essential git libusb-1.0-0 libusb-1.0-0-dev`

See more details here:  https://github.com/node-hid/node-hid#compiling-from-source

If that doesn't work, please see if you can get a better output of `npm install node-hid`.

tbnobody, 2019-02-09:
I have a similar problem. 
I just ran the install with `npm install --verbose node-hid --build-from-source`

info using npm@6.5.0
npm info using node@v11.9.0

And got:

```
../src/HID.cc: In member function ‘void HID::readResultsToJSCallbackArguments(HID::ReceiveIOCB*, v8::Local<v8::Value>*)’:
../src/HID.cc:207:74: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
     Local<Object> buf = nodeBufConstructor->NewInstance(1, nodeBufferArgs);
```
I am using the lastest Rasbian Version:
```
cat /etc/debian_version
9.6
```

todbot, 2019-02-10:
Hi,
How are you getting "9.6" for Raspbian version?  The latest version on raspberrypi.org is 9.4.

I just installed `node-hid` on a fresh install of Raspian.  The process is documented here:
 https://gist.github.com/todbot/ee76690bc0da93203ffb67a60cf9cafc

tbnobody, 2019-02-10:
I got version 9.6 by updating 9.4 version (apt update && apt upgrade). Another remark, I downgraded nodejs to version 8.15.0 (latest version in 8 branch) and it worked perfectly.

closed 2019-02-14 by todbot

chepe263, 2020-07-24:
Just a side note for future users
In `Raspbian GNU/Linux 10 (buster)`, it was necesarry to install two packages using
`sudo apt install libusb-1.0-0-dev libudev-dev`