node-hid/node-hid #300: Update table of supported Electron version

robatwilliams, 2019-03-08:
The table only goes up to 1.8

Comments here confirm it works with 4.0.4 https://github.com/node-hid/node-hid/issues/293#issuecomment-464247084

and I can confirm that it works with 4.0.7.

jschimmoeller, 2019-05-03:
i have validated that it does work with 4.x.x

todbot, 2019-05-03:
It actually works up to Node v12 and Electron v5.  
The table in the README needs to be redesigned. :)

haakonnessjoen, 2019-06-08:
How do you get it to compile with Electron v5?

I am getting:
```../src/HID.cc:207:45: error: no matching member function for call to 'NewInstance'
    Local<Object> buf = nodeBufConstructor->NewInstance(1, nodeBufferArgs);
                        ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~```

closed 2019-06-08 by todbot