node-hid/node-hid #302: cannot open device with path Symbol scanner

cvinodhkumar77, 2019-04-08:
I trying to get data from Symbol scanner device from Electron container.  I get below error when try with the code

```
if( deviceInfo ) {
  var device = new HID.HID( deviceInfo.path ); //Error here
  // ... use device
}
```
OS: Windows 10

> cannot open device with path \\?\hid#vid_05e0&pid_1200#6&985766a&1&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

todbot, 2019-04-08:
Is the scanner device acting like a USB HID keyboard?  If so, you cannot access it with `node-hid` because the Windows OS claims it, for security reasons (otherwise you could build a keylogger with any program that has access to the HID APIs)

cvinodhkumar77, 2019-04-08:
This might be irrelevant here but how can I find that my device is acting like a USB HID keyboard.  If I open notepad and scan something I can see the data populated on notepad.  Does that mean my scanner device acts like a USB HID keyboard?

todbot, 2019-04-08:
Yes, if it acts like a keyboard to Notepad, then the OS has no way to distinguish between it and a regular keyboard.

Some scanner devices have a switch to alternate between USB modes, like:  USB HID Keyboard, USB HID Data, USB Serial.  If you can switch the device to USB HID Data mode, then you can use  `node-hid`. If you can switch it to USB Serial mode, you can use `node-serial`.

cvinodhkumar77, 2019-04-09:
Closing as I was able to change the device mode and used serial port for getting data

closed 2019-04-09 by cvinodhkumar77