node-hid/node-hid #307: Unable to communicate with raw HID device on Windows 10

PTS93, 2019-05-02:
I'm talking to an nRF52840 (using Arduino) via node-hid.
This is working fine on Ubuntu 18.04, where I can send data back and forth.

On Windows 10 on the other hand this is not working. The device manager does not show the device as a keyboard or mouse so I'm assuming this is not hitting the known issue with those devices. Rather it is shown as a "HID-compliant vendor defined device" e.g. raw HID (afaik).

So the same node.js script with the same device works fine on linux but seems stuck on windows. After `console.log(deviceInfo)` nothing seems to be happening.

```js
var HID = require('node-hid');
var devices = HID.devices();
var deviceInfo = devices.find( function(d) {
    var isNRF = d.vendorId===0X239A && d.productId===0X8029;
    return isNRF;
});
if( deviceInfo ) {
	console.log(deviceInfo)
	var device = new HID.HID( deviceInfo.path );
	device.on("data", function(data) {console.log(data)});

	device.on("error", function(err) {console.log(err)});

	setInterval(function () {
		device.write([0x00, 0x01, 0x01, 0x05, 0xff, 0xff]);
	},500)
}
```

todbot, 2019-05-02:
Depending on the HID descriptor of your device, you will probably need to prepend a zero byte to the data you write to your device.
See: https://github.com/node-hid/node-hid#prepend-byte-to-hid_write

PTS93, 2019-05-02:
I tried the following already but the outcome is the same.
```js
	var buffer = Array(64).fill(0x33); // device has 64-byte report
	buffer.unshift(0);  // prepend throwaway byte

	setInterval(function () {
		device.write(buffer);
	},500)
```

todbot, 2019-05-02:
Two questions:
1. Are you creating an array/buffer that is the exact size (or size+1) of the HID report you're sending? 
2. What version of Node are you using?  If you're using Node v12 and `node-hid@0.7.8`, try going back to `node-hid@0.7.7` and Node v10 or similar.  I just released a node-hid update and I'm not entirely sure the changes made work for all cases.

PTS93, 2019-05-02:
At the moment the "Arduino" is just echoing what it received.
I'm using node 10.12. I downgraded to node-hid@0.7.7 but the result is the same.

PTS93, 2019-05-02:
The output of deviceinfo is relevant:
```
 vendorId: 9114,
  productId: 32809,
  path:
   '\\\\?\\hid#vid_239a&pid_8029&mi_02#....',
  serialNumber: 'C78A1CBBD32DC15D',
  manufacturer: 'Adafruit Industries',
  product: 'Bluefruit nRF52840',
  release: 256,
  interface: 2,
  usagePage: 65280,
  usage: 1 }
```

todbot, 2019-05-02:
Do you have a link to the Arduino sketch you're using (and what Board you've set the IDE to)?  I may be able to replicate your setup.

PTS93, 2019-05-02:
Sure!
You will need an nRF52840 device though.
If you do have a board with that chip you will need to do the following to replicate the setup:

- Download the master of https://github.com/adafruit/Adafruit_nRF52_Arduino
And follow the instructions for "Adafruit nRF52 BSP via git" in the readme
- Copy and replace the `src` directory of this repo https://github.com/hathach/tinyusb/blob/master/hw/bsp/metro_m0_express/board.mk with the one located at `Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino\cores\nRF5\Adafruit_TinyUSB_core\tinyusb`
- Download this library into your libraries folder https://github.com/adafruit/Adafruit_TinyUSB_Arduino

Here also the relevant issue to this development: https://github.com/hathach/tinyusb/issues/58

If you dont have an nRF52840 device you can probably replicate the same behaviour with the "Arduino free" version of this example located here: https://github.com/hathach/tinyusb/tree/master/examples/device/hid_generic_inout

It should support all devices listed as supported by tinyusb.

PTS93, 2019-05-02:
Ah sorry and the actual sketch is then: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/hid_generic_inout/hid_generic_inout.ino

todbot, 2019-05-02:
I'll take a poke around my shop and see which boards I have can do TinyUSB. 

Another question: what is the output of `HID.devices()` on both Linux and Windows?

PTS93, 2019-05-02:
Windows:
```
[ { vendorId: 6940,
    productId: 6921,
    path:
     '\\\\?\\hid#vid_1b1c&pid_1b09...}',
    manufacturer: 'Corsair',
    product: 'Corsair K70B Gaming Keyboard',
    release: 265,
    interface: 0,
    usagePage: 1,
    usage: 6 },
  { vendorId: 6940,
    productId: 6921,
    path:
     '\\\\?\\hid#vid_1b1c&pid_1b09...}',
    manufacturer: 'Corsair',
    product: 'Corsair K70B Gaming Keyboard',
    release: 265,
    interface: 2,
    usagePage: 1,
    usage: 6 },
  { vendorId: 6940,
    productId: 6921,
    path:
     '\\\\?\\hid#vid_1b1c&pid_1b09...}',
    manufacturer: 'Corsair',
    product: 'Corsair K70B Gaming Keyboard',
    release: 265,
    interface: 1,
    usagePage: 12,
    usage: 1 },
  { vendorId: 2362,
    productId: 9505,
    path:
     '\\\\?\\hid#vid_093a&pid_2521...',
    product: 'USB OPTICAL MOUSE',
    release: 256,
    interface: -1,
    usagePage: 1,
    usage: 2 },
  { vendorId: 9114,
    productId: 32809,
    path:
     '\\\\?\\hid#vid_239a&pid_8029...}',
    serialNumber: 'C78A1CBBD32DC15D',
    manufacturer: 'Adafruit Industries',
    product: 'Bluefruit nRF52840',
    release: 256,
    interface: 2,
    usagePage: 65280,
    usage: 1 } ]
```

Linux:
```
[ { vendorId: 9114,
    productId: 32809,
    path: '/dev/hidraw2',
    serialNumber: 'C78A1CBBD32DC15D',
    manufacturer: 'Adafruit Industries',
    product: 'Bluefruit nRF52840',
    release: 256,
    interface: 2,
    usagePage: 67 } ]
```

todbot, 2019-05-02:
Hmm.. okay so it's not showing up as two devices with different usages (which is something I've seen before).

One last thing to try: change your `Array(64)` to `Array(16)`.  Without looking at the HID report descriptor (and with my very small understanding how TinyUSB is being used), it's hard to tell how much data you should actually be sending and if you should be sending a report descriptor or not.   It looks like it's either 64 bytes or 16 bytes, and no report descriptor.

So try 16 instead of 64. :)

todbot, 2019-05-03:
I am trying to follow the steps you outlined above but I think the second step should be:
> Copy and replace the `src` directory of this repo https://github.com/hathach/tinyusb/
with the `src` directory at: `Arduino\hardware\Adafruit\Adafruit_nRF52_Arduino\cores\nRF5\Adafruit_TinyUSB_core\tinyusb`
but after doing all those steps and trying to compile the Blink sketch, I get a bunch of linker errors since it can't find `tud_hid_get_report_cb`, `tud_hid_set_report_cb`, and similar TinyUSB functions. Is there another step?  I shouldn't think so since you're just swapping out TinyUSB for another one or are is the fact you're referencing a .mk file significant?

PTS93, 2019-05-03:
This only work if you compile for an nRF52840 target. If choose for example the Feather nRF52840 Express it should compile, otherwise it will not link to tinyusb.

todbot, 2019-05-03:
Yup, I'm set to the "Adafruit Bluefruit nRF52840 Feather Express" board.

PTS93, 2019-05-03:
Did you maybe install the nRF52 core from the board manager at some point? If there is a core in the appdata/local/.arduino15 folder it will use that instead of the one located in the Documents/Arduino directory.

todbot, 2019-05-03:
Okay I got it working now.  I think some cache was slightly corrupt because I re-added the core and TinyUSB and unticked "aggressively cache compiled core" and now it compiles (albeit with some warnings I assume are expected)  Now I hopefully I can start some tests on hardware.

Did you try the 64 -> 16 byte change?

PTS93, 2019-05-03:
So far no, I would try that out as soon as I get home.

todbot, 2019-05-03:
Another avenue to investigate: modify the Arduino sketch (if need be) so it is compatible with the RawHID executables that PJRC supplies for Teensy:
  https://www.pjrc.com/teensy/rawhid.html
The Arduino sketch he uses for this (which I couldn't find on github), I put here:
  https://gist.github.com/todbot/4697892e742b1d02fd11369842f405e6

PTS93, 2019-05-03:
So setting the buffer size to 16 didn't change anything, then again the original example was only 6 bytes already.
One would probably have to look at the USB implementation from teensy to determine the root cause. The sketch itself doesn't seem to tell much.
So the teensy example is still working in the latest version?

hathach, 2019-05-06:
Ok, I just run the following script
 
```
var HID = require('node-hid');
var devices = HID.devices();
var deviceInfo = devices.find( function(d) {
    var isNRF = d.vendorId===0X239A && d.productId===0X8029;
    return isNRF;
});
if( deviceInfo ) {
	console.log(deviceInfo)
	var device = new HID.HID( deviceInfo.path );
	device.on("data", function(data) {console.log(data)});

	device.on("error", function(err) {console.log(err)});

	setInterval(function () {
		device.write([0x00, 0x01, 0x01, 0x05, 0xff, 0xff]);
	},500)
}
```
output on ubuntu

```
$ node hid.js 
{ vendorId: 51966,
  productId: 16388,
  path: '/dev/hidraw8',
  serialNumber: '123456',
  manufacturer: 'tinyusb.org',
  product: 'tinyusb device',
  release: 256,
  interface: 0,
  usagePage: 14901,
  usage: 11825 }
<Buffer 01 01 05 ff ff>
<Buffer 01 01 05 ff ff>
<Buffer 01 01 05 ff ff>
<Buffer 01 01 05 ff ff>
<Buffer 01 01 05 ff ff>
```

on windows

```

```

todbot, 2019-12-05:
Does this work now, with latest node-hid?

PTS93, 2019-12-06:
I haven't tested it in a while but since hatach implemented the changes in TinyUSB it was all working. I would close this for now.

closed 2019-12-06 by PTS93