node-hid/node-hid #308: Wrong Module Version in 0.7.8 v70 prebuilds

Kasalehlia, 2019-05-09:
Hello,
node-hid-v0.7.8-electron-v70-win32-x64.tar.gz actually contains `HID.node` with `NODE_MODULE_VERSION` 47, so Electron 5.0.1 does not load this binding.

todbot, 2019-05-09:
Hi,
Confirmed.  There were issues with `node-abi` and `prebuild` but I think they've been dealt with. I've updated the prebuilds for Windows and MacOS for Electron v5.  Let me know if it works.

Note, you'll likely need to delete the previous prebuilds from your cache, see https://github.com/prebuild/prebuild-install#cache for details.  But for Windows, just delete the directory `$ENV:APPDATA\npm-cache\_prebuilds`

closed 2019-05-09 by todbot

Kasalehlia, 2019-05-10:
Works perfectly, thank you!