node-hid/node-hid #309: linux libusb support

Julusian, 2019-05-29:
It has been a while since the problems in #242 were found, so I have had a play to see if libusb is still blocked on the same problems.

With some minimal mods to build HID-libusb.node, it appears to just work for me. Am I missing something in what is really wrong, or does this mean it now works?

The steps followed:
1. `git checkout v0.7.8`
1. `npm install`
1. Add to bindings.gyp (probably breaks other platforms)
	```
	{
      'target_name': 'HID-libusb',
      'sources': [ 'src/HID.cc' ],
      'dependencies': ['hidapi-libusb'],
      'defines': [
        '_LARGEFILE_SOURCE',
        '_FILE_OFFSET_BITS=64',
      ],
      'conditions': [
        [ 'OS=="linux"', {
          'libraries': [
            '-lusb-1.0'
          ]
        }]
      ],
      'cflags!': ['-ansi', '-fno-exceptions' ],
      'cflags_cc!': [ '-fno-exceptions' ],
      'cflags': ['-g', '-exceptions'],
      'cflags_cc': ['-g', '-exceptions']
    },
    {
      'target_name': 'hidapi-libusb',
      'type': 'static_library',
      'conditions': [
        [ 'OS=="linux"', {
          'sources': [ 'hidapi/libusb/hid.c' ],
          'include_dirs+': [
            '/usr/include/libusb-1.0/'
          ]
        }]
      ],
      'direct_dependent_settings': {
        'include_dirs': [
          'hidapi/hidapi',
          "<!(node -e \"require('nan')\")"
        ]
      },
      'include_dirs': [
        'hidapi/hidapi'
      ],
      'defines': [
        '_LARGEFILE_SOURCE',
        '_FILE_OFFSET_BITS=64',
      ],
      'cflags': ['-g'],
      'cflags!': [
        '-ansi'
      ]
    }
	```
1. `./node_modules/.bin/prebuild -t 8.16.0`
1. `cp prebuilds/node-hid-v0.7.8-node-v57-linux-x64.tar.gz ~/.npm/_prebuilds/c457b7-node-hid-v0.7.8-node-v57-linux-x64.tar.gz` to inject into the npm prebuild cache
1. In a test project `npm install` and modify node_modules/node-hid/nodehid.js with the following
	```
	function loadBinding() {
	    if( !binding ) {
	        if( os.platform() === 'linux' ) {
	            // Linux defaults to hidraw
	            if( !driverType || driverType === 'libusb' ) {
	                binding = require('bindings')('HID-libusb.node');
	            } else {
	                binding = require('bindings')('HID.node');
	            }
	        }
	        else {
	            binding = require('bindings')('HID.node');
	        }
	    }
	}
	```

todbot, 2019-05-29:
The problem as it stood back when I last looks was that `prebuild` had no support for multiple native files per platform. So we could either distribute Linux prebuilds with the hidraw driver or the libusb driver but not both.

todbot, 2019-05-29:
Oh, looks like there was further movement on https://github.com/prebuild/prebuild/issues/169 and it may be fixed. Looking into it now...

Julusian, 2019-05-29:
OK, from my test I think that it does now work. As far as I can tell I followed the flow the files would have gone exactly, except for the http transfers but that wouldnt affect the file contents.

Thanks, it would be great to have this fixed :)

todbot, 2019-05-29:
Testing it now and yes it appears both `prebuild` can p)ut in multiple `.node` files now and `prebuild-install` will pull them both down.

What does libusb give you that hidraw doesn't?  (Just curious. There was a big push to move HID stuff to hidraw because of the deficiencies of libusb

Julusian, 2019-05-29:
Thats great to hear

I am trying to use https://github.com/Lange/node-elgato-stream-deck
It has already been mentioned here a while ago https://github.com/node-hid/node-hid/issues/249

todbot, 2019-05-29:
Oh right, that hidraw report size limit!  I mean they're not wrong that it's not what HID is for, but still :)
Hopefully I'll have node-hid@0.7.9 up later tonight. Thanks for the help!

Julusian, 2019-05-31:
I can confirm that the 0.7.9 release done works well.
Thanks for the rapid response on resolving this

closed 2019-05-31 by Julusian