node-hid/node-hid #312: Win 10 1903 Blocks USB HID

onlykey, 2019-06-07:
In the latest version of Windows USB HID devices are only accessible if your app is run as administrator. I am not sure why Windows did this but I found some verification here - https://groups.google.com/a/fidoalliance.org/d/msg/fido-dev/iDLKY-HsY0A/4ibOVQaUBAAJ

Any ideas on workaround here? I found that even .open(vendorid, productid) is not working.

todbot, 2019-06-07:
That post seems to be talking about WebAPIs, not USB or HID APIs.
I am currently updating one of my test VMs to see if I can observe any issue.
If this is an issue, it's bigger than node-hid, that's for sure.

todbot, 2019-06-08:
Okay I've checked on a Windows 10 Pro build 1903 system, both as an Administrator user and a normal user and USB HID does not seem to be impacted.  

@onlykey do you have any other details on what issue you're seeing?  I cannot replicate so I am closing this for now.

closed 2019-06-08 by todbot

onlykey, 2019-06-12:
@todbot The link was about Webauthn but that uses USB HID too, and using USB HID doesn't work for Windows 10 1903 with these devices without admin now.

I did another test using the Teensy raw hid example, and you are right it works fine, thanks for testing this. It seems the issue is less broad than I thought, Win 10 1903 blocks USB HID devices only if they have a usage page of 0xF1D0 and usage of 1 (Settings to use for a FIDO U2F or Webauthn device). These settings are changed in usb_desc.h of teensy3 core. 

Here is the teensy sketch with usage page 0xF1D0 - 
[Basic.pde.hexf1d0.zip](https://github.com/node-hid/node-hid/files/3281525/Basic.pde.hexf1d0.zip)
^^^ Only works if run as admin

Here is the teensy sketch with default teensy settings - 
[Basic.pde.hex.zip](https://github.com/node-hid/node-hid/files/3281475/Basic.pde.hex.zip)
^^^ Works as regular user

Have to change the test-teensyrawhid.js to these settings to test:

    return ((d.usagePage===0xF1D0 && d.usage===1) || (d.usagePage===0xF1D0 && d.usage===1));

I guess since this is Windows intentionally blocking these USB HID devices there is not anything that can be changed in hidapi. Apps won't be able to use node-hid to communicate with U2F/FIDO2 devices in Windows.