node-hid/node-hid #313: Add support for electron 6

setpixel, 2019-06-10:
hi! 

I'd like to support hid devices for an electron 6 app.

I currently get: 

```
Error: /usr/local/bin/node exited with code 1
Output:

> node-hid@0.7.9 install /Users/setpixel/git/storyboarder/node_modules/node-hid
> prebuild-install || node-gyp rebuild

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
  SOLINK_MODULE(target) Release/HID.node

> fsevents@1.2.7 install /Users/setpixel/git/storyboarder/node_modules/fsevents
> node install

  SOLINK_MODULE(target) Release/.node
  CXX(target) Release/obj.target/fse/fsevents.o
Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/Users/setpixel/git/storyboarder/node_modules/fsevents/lib/binding/Release/electron-v6.0-darwin-x64/fse.node --module_name=fse --module_path=/Users/setpixel/git/storyboarder/node_modules/fsevents/lib/binding/Release/electron-v6.0-darwin-x64 --napi_version=4 --node_abi_napi=napi' (1)

Error output:
/Users/setpixel/git/storyboarder/node_modules/node-abi/index.js:30
  throw new Error('Could not detect abi for version ' + target + ' and runtime ' + runtime + '.  Updating "node-abi" might help solve this issue if it is a new release of ' + runtime)
  ^

Error: Could not detect abi for version 6.0.0-beta.6 and runtime electron.  Updating "node-abi" might help solve this issue if it is a new release of electron
    at getAbi (/Users/setpixel/git/storyboarder/node_modules/node-abi/index.js:30:9)
    at module.exports (/Users/setpixel/git/storyboarder/node_modules/prebuild-install/rc.js:73:57)
    at Object.<anonymous> (/Users/setpixel/git/storyboarder/node_modules/prebuild-install/bin.js:9:25)
    at Module._compile (internal/modules/cjs/loader.js:759:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:770:10)
    at Module.load (internal/modules/cjs/loader.js:628:32)
    at Function.Module._load (internal/modules/cjs/loader.js:555:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:826:10)
    at internal/main/run_main_module.js:17:11
node-pre-gyp WARN Using request for node-pre-gyp https download 
In file included from ../fsevents.cc:6:
In file included from ../../nan/nan.h:222:
In file included from ../../nan/nan_converters.h:67:
```

todbot, 2019-06-10:
Electron v6 has not been released yet. `electron-rebuild`  depends on `node-abi`, which tracks current valid Electron and Node releases.  Once they both get updated (hopefully before Electron v6 is released), `node-hid` will install cleanly.

There is no work-around as far as I know.  If you need the functionality of Electron v6, I'd suggest stubbing out your `node-hid` calls for now.  Otherwise, use Electron v5.

danoespinoza, 2019-07-30:
Electron v6 has been released and get Error: Could not detect abi for version 6.0.0 and runtime electron.

todbot, 2019-07-30:
You should take this up with `electron-rebuild`.  As mentioned above, `node-hid` depends on it and it depends on `node-abi`.  `electron-rebuild` hasn't been updated yet for Electron v6.

todbot, 2019-07-30:
Okay it looks like `node-abi` was updated 20 minutes ago to support Electron v6.  If you explicitly add the latest version of `node-abi` then `node-hid` will compile. E.g. change your project's `devDependencies` from this:
```
  "devDependencies": {
    "electron": "^6.0.0",
    "electron-rebuild": "^1.8.5"
  }
```

to this:

```
  "devDependencies": {
    "electron": "^6.0.0",
    "electron-rebuild": "^1.8.5",
    "node-abi": "^2.10.0"
  }
```

cumajkeee, 2019-08-01:
Hi @todbot, after added abi to dev dependencies got following:
```
prebuild-install http request GET https://github.com/node-hid/node-hid/releases/download/v0.7.9/node-hid-v0.7.9-electron-v73-darwin-x64.tar.gz
prebuild-install http 404 https://github.com/node-hid/node-hid/releases/download/v0.7.9/node-hid-v0.7.9-electron-v73-darwin-x64.tar.gz

```

todbot, 2019-08-01:
Yes, there are not yet `node-hid` prebuilt binaries yet for Electron v6 so you will have to let it compile on your system.

orangeChu, 2019-10-28:
Electron 6 is now supported. However, electron7 is not supported.
cc @todbot

closed 2019-11-23 by todbot