node-hid/node-hid #316: Getting node-hid to work in electron (Could not locate the bindings file)

sontek, 2019-06-21:
I'm currently can't find the bindings file, which I've seen many times in the issues but it always says you run rebuild:

```
×
Error: Could not locate the bindings file. Tried:
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\build\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\build\Debug\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\build\Release\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\out\Debug\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\Debug\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\out\Release\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\Release\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\build\default\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\compiled\12.0.0\win32\x64\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\addon-build\release\install-root\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\addon-build\debug\install-root\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\addon-build\default\install-root\HID.node
 → C:\Users\sonte\Documents\GitHub\eventray_onsite\lib\binding\node-v70-win32-x64\HID.node
```

But as you can see in my package.json I have told it to rebuild:

```
    "postinstall": "electron-rebuild -f -w node-hid && electron-builder install-app-deps",
```

and you can see when I run yarn it does rebuild:
```
PS C:\Users\sonte\Documents\GitHub\eventray_onsite> yarn
yarn install v1.12.3
info No lockfile found.
[1/4] Resolving packages...
warning react-scripts > fsevents@2.0.6: Please update: there are crash fixes
warning react-scripts > postcss-preset-env > postcss-color-functional-notation > postcss-values-parser > flatten@1.0.2: I wrote this module a very long time ago; you should use something else.
warning react-scripts > jest > jest-cli > jest-config > jest-environment-jsdom > jsdom > left-pad@1.3.0: use String.prototype.padStart()
warning react-scripts > webpack > micromatch > snapdragon > base > mixin-deep@1.3.1: Critical bug fixed in v2.0.1, please upgrade to the latest version.
warning react-scripts > webpack > micromatch > snapdragon > base > cache-base > set-value@2.0.0: Critical bug fixed in v3.0.1, please upgrade to the latest version.
warning react-scripts > webpack > micromatch > snapdragon > base > cache-base > union-value > set-value@0.4.3: Critical bug fixed in v3.0.1, please upgrade to the latest version.
warning wait-on > joi@13.7.0: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
warning wait-on > joi > hoek@5.0.4: This version has been deprecated in accordance with the hapi support policy (hapi.im/support). Please upgrade to the latest version to get the best features, bug fixes, and security patches. If you are unable to upgrade at this time, paid support is available for older versions (hapi.im/commercial).
warning wait-on > joi > topo@3.0.3: This module has moved and is now available at @hapi/topo. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.
warning wait-on > joi > topo > hoek@6.1.3: This module has moved and is now available at @hapi/hoek. Please update your dependencies as this version is no longer maintained an may contain bugs and security issues.
warning electron > electron-download > nugget > progress-stream > through2 > xtend > object-keys@0.4.0:
[2/4] Fetching packages...
info fsevents@2.0.6: The platform "win32" is incompatible with this module.
info "fsevents@2.0.6" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@1.2.9: The platform "win32" is incompatible with this module.
info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
warning " > @rescripts/cli@0.0.10" has incorrect peer dependency "react-scripts@^2.1.2".
warning "@rescripts/cli > @rescripts/utilities@0.0.5" has incorrect peer dependency "react-scripts@^2.1.2".
warning "@rescripts/rescript-env > @rescripts/rescript-use-tslint-config > tslint-loader@3.5.4" has unmet peer dependency "tslint@>=4.0.0".
[4/4] Building fresh packages...
success Saved lockfile.
$ electron-rebuild -f -w node-hid && electron-builder install-app-deps
√ Rebuild Complete
  • electron-builder version=20.44.4
  • loaded configuration file=package.json ("build" field)
  • rebuilding native production dependencies platform=win32 arch=x64
  • rebuilding native dependency name=node-hid
  • rebuilding native dependency name=node-sass
Done in 106.64s.

```

Any idea why I'd still be getting this error?

sontek, 2019-06-21:
Thats the error if I try to show it in the renderer, if I do it in the main electron side I get that node module versions are different:

[image]

todbot, 2019-06-22:
Hi @sontek, 
It sounds like your build system isn't running `electron-rebuild` correctly.
Two things I would try:

1. Change your "postinstall" to just `"electron-rebuild"` and explicitly do a "`npm run postinstall`" (or equivalent) before launching your app.

2. Can you download and run https://github.com/todbot/electron-hid-toy ?  This is a minimal Electron app using `node-hid`.  This might show if there's any issues with your Node installation.

Robert-Baindorov-Mindtree, 2019-07-05:
I have the same problem and here is the output from the electron-hid-toy:

robert@pop-os:~/bose_project/electron-hid-toy$ npm install 
npm WARN deprecated core-js@1.2.7: core-js@<2.6.8 is no longer maintained. Please, upgrade to core-js@3 or at least to actual version of core-js@2.

> node-hid@0.7.9 install /home/robert/bose_project/electron-hid-toy/node_modules/node-hid
> prebuild-install || node-gyp rebuild

> core-js@2.6.9 postinstall /home/robert/bose_project/electron-hid-toy/node_modules/core-js
> node scripts/postinstall || echo "ignore"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon: 
> https://opencollective.com/core-js 
> https://www.patreon.com/zloirock 

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> electron@5.0.6 postinstall /home/robert/bose_project/electron-hid-toy/node_modules/electron
> node install.js

Downloading tmp-17529-1-SHASUMS256.txt-5.0.6
[============================================>] 100.0% of 4.74 kB (4.74 kB/s)

> electron-hid-toy@0.1.11 postinstall /home/robert/bose_project/electron-hid-toy
> npm run rebuild

> electron-hid-toy@0.1.11 rebuild /home/robert/bose_project/electron-hid-toy
> electron-rebuild

✔ Rebuild Complete
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.2.9: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

added 691 packages from 541 contributors and audited 3418 packages in 27.706s
found 1 low severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details
robert@pop-os:~/bose_project/electron-hid-toy$ npm start

> electron-hid-toy@0.1.11 start /home/robert/bose_project/electron-hid-toy
> electron . --enable-logging --remote-debugging-port=9222

App threw an error during load
Error: The module '/home/robert/bose_project/electron-hid-toy/node_modules/node-hid/build/Release/HID-hidraw.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 47. This version of Node.js requires
NODE_MODULE_VERSION 70. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar.js:155:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:857:18)
    at Object.func [as .node] (electron/js2c/asar.js:155:31)
    at Module.load (internal/modules/cjs/loader.js:677:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:609:12)
    at Function.Module._load (internal/modules/cjs/loader.js:601:3)
    at Module.require (internal/modules/cjs/loader.js:715:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at bindings (/home/robert/bose_project/electron-hid-toy/node_modules/bindings/bindings.js:112:48)
    at loadBinding (/home/robert/bose_project/electron-hid-toy/node_modules/node-hid/nodehid.js:19:46)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: The module '/home/robert/bose_project/electron-hid-toy/node_modules/node-hid/build/Release/HID-hidraw.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 47. This version of Node.js requires
NODE_MODULE_VERSION 70. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar.js:155:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:857:18)
    at Object.func [as .node] (electron/js2c/asar.js:155:31)
    at Module.load (internal/modules/cjs/loader.js:677:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:609:12)
    at Function.Module._load (internal/modules/cjs/loader.js:601:3)
    at Module.require (internal/modules/cjs/loader.js:715:19)
    at require (internal/modules/cjs/helpers.js:14:16)
    at bindings (/home/robert/bose_project/electron-hid-toy/node_modules/bindings/bindings.js:112:48)
    at loadBinding (/home/robert/bose_project/electron-hid-toy/node_modules/node-hid/nodehid.js:19:46)

DevTools listening on ws://127.0.0.1:9222/devtools/browser/28e53ffd-825e-4090-ae1a-189a29eaf89c

todbot, 2019-07-05:
Hi @Robert-Baindorov-Mindtree,
What OS are you on?

todbot, 2019-07-05:
It looks like you're on either Linux or MacOS, so try this:
```
rm -rf ~/.npm/_prebuilds
cd electron-hid-toy
rm -rf node_modules
npm install
npm run start
```

What's going on here is there has been some issues with the `prebuild` and `node-gyp` with Electron v5+.  And `node-hid` uses those modules.  I think I cleaned it up, but it does mean you'll need to delete any previously downloaded prebuild.

todbot, 2019-07-05:
Oh and @sontek, for Windows you should delete the folder `%APPDATA%\npm-cache` and try re-downloading electron-hid-toy and reinstalling.

Robert-Baindorov-Mindtree, 2019-07-08:
Amazing, that worked! Thank you so much! 
Just for others reference:
I am on Pop_OS from System76. Basically Ubuntu Cosmic. 

```
robert@pop-os:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Pop!_OS 18.10
Release:	18.10
Codename:	cosmic
robert@pop-os:~$ 

robert@pop-os:~$ node --version
v10.16.0
robert@pop-os:~$ npm --version
6.9.0
```

Robert-Baindorov-Mindtree, 2019-07-08:
Unfortunately I am still getting errors in the main project that I am trying to do development on.

So sorry, I know this is not your problem, but here is the error:

```
New Instance of HID is created !!
App threw an error during load
Error: cannot open device with vendor id 0x1d6b and product id 0x101
    at new HID (/home/robert/bose_project/Raphael-App-Framework/node_modules/node-hid/nodehid.js:49:17)
    at Object.getDeviceHandle (/home/robert/bose_project/Raphael-App-Framework/server/services/utilities.service.js:21:28)
    at new DeviceController (/home/robert/bose_project/Raphael-App-Framework/server/controllers/device.controller.js:21:70)
    at Object.<anonymous> (/home/robert/bose_project/Raphael-App-Framework/server/controllers/device.controller.js:172:32)
    at Module._compile (internal/modules/cjs/loader.js:808:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:820:10)
    at Module.load (internal/modules/cjs/loader.js:677:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:609:12)
    at Function.Module._load (internal/modules/cjs/loader.js:601:3)
    at Module.require (internal/modules/cjs/loader.js:715:19)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: cannot open device with vendor id 0x1d6b and product id 0x101
    at new HID (/home/robert/bose_project/Raphael-App-Framework/node_modules/node-hid/nodehid.js:49:17)
    at Object.getDeviceHandle (/home/robert/bose_project/Raphael-App-Framework/server/services/utilities.service.js:21:28)
    at new DeviceController (/home/robert/bose_project/Raphael-App-Framework/server/controllers/device.controller.js:21:70)
    at Object.<anonymous> (/home/robert/bose_project/Raphael-App-Framework/server/controllers/device.controller.js:172:32)
    at Module._compile (internal/modules/cjs/loader.js:808:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:820:10)
    at Module.load (internal/modules/cjs/loader.js:677:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:609:12)
    at Function.Module._load (internal/modules/cjs/loader.js:601:3)
    at Module.require (internal/modules/cjs/loader.js:715:19)
```

todbot, 2019-07-08:
Hi  @Robert-Baindorov-Mindtree,
That looks like a udev permssions issue.   Try following what's described here:
https://github.com/node-hid/node-hid#udev-device-permissions

sontek, 2019-07-10:
working for me now

closed 2019-07-10 by sontek