node-hid/node-hid #319: node-hid + electron + typescript

warrick-eosny, 2019-07-17:
I want to develop my Electron app in typescript I've yanked this example project and added node-hid to it https://github.com/wwlib/cra-craco-electron-example and adapted it for two package structure. 

I'm running myself in circles though. 
In app\electron-dev.js (the main entrypoint) this code works fine. 

```
var HID = require('node-hid');
console.log("node-hid loaded: devices:", HID.devices());
```

but in my react application, in src\model\TestUSB.ts I have code that looks like this:

```
let HID: any;
if (process.env.REACT_APP_MODE === 'electron') {
    console.log(`REQUIRING node-hid`);
    HID = require('node-hid');
    // console.log("node-hid loaded: devices:", HID.devices());
    console.log("node-hid loaded: devices: DUMMY");
}

export default class TestFs {

    static getDirectoryListing(): string {
        if (process.env.REACT_APP_MODE === 'electron') {
            let usbDevices = HID.devices();
            return JSON.stringify(usbDevices, null, 2);
        } else {
            return 'USB device lising is not available in the browser.'
        }
    }
}
```

I keep running into this error. 

[image]

While running in circles I've tried running

```
npx electron-rebuild –force
```
Which I did find resolved this problem for me in the other cases, but for some reason does not solve the problem in this case?

My project can be found here: https://github.com/warrick-eosny/electron-typescript-node-hid

Steps:
1. Clone repo
2. yarn watch:electron
3. yarn start:electron-dev

todbot, 2019-07-18:
Unfortunately I'm not familiar with Typescript.  I'll take a poke at your project.
That error is what you see when Electron is trying to load a native-module built for Node.
That's what you see when `electron-rebuild` isn't run.

todbot, 2019-07-18:
I'm also not very familiar with `yarn`.  Do you not have to do the equivalent of `npm install` after the clone?

todbot, 2019-07-18:
Well, I think I got it to work, I think.  After `yarn install`, I tried watch & start and saw it fail. Then I did:
- stopped all yarn watch / yarn start processes 
- removed `node-hid` from `./package.json` (it's part of the app not the build process)
- `rimraf node_modules/node-hid`  
- `./node_modules/.bin/electron-rebuild -w node-hid -f`
- `yarn watch:electron` (in one window)
- `yarn start:electron-dev` (in another window)

warrick-eosny, 2019-07-18:
mmm that's interesting. So I'd added node-hid to the ./package.json file because I was getting this error:

[image]

And after following your steps I am back at this point again. 

The only thing I've noticed is that you're doing this on a mac / linux box and I'm on Windows. I'm going to try the same test on my Mac for good measure also (but I don't think that has much to do with the issue)

So on windows this was the exact set of steps I took
```
git clone https://github.com/warrick-eosny/electron-typescript-node-hid.git clean2                          
cd .\clean2\                                                                                                
yarn                                                                                                        
del .\node_modules\node-hid\                                                                                
.\node_modules\.bin\electron-rebuild -w node-hid -f                                                         
yarn watch:electron (in one window)
yarn start:electron-dev (in another window)
```

I should also mention that I copied the webpack rule you have in the Blink1Control2 project to make sure that I don't package node-hid with webpack also. 
https://github.com/todbot/Blink1Control2/blob/master/webpack.config.js

Will report back on my Mac test shortly

warrick-eosny, 2019-07-18:
No I get the same result on my Mac ... 

You must have some secret sauce there :) 

[image]

```
git clone https://github.com/warrick-eosny/electron-typescript-node-hid.git clean2 
cd clean2/
yarn
rimraf node_modules/node-hid
rm -Rf node_modules/node-hid
./node_modules/.bin/electron-rebuild -w node-hid -f
yarn watch:electron
```

tahesse, 2020-01-22:
> I want to develop my Electron app in typescript I've yanked this example project and added node-hid to it https://github.com/wwlib/cra-craco-electron-example and adapted it for two package [...]

This makes sense because this part of electron is the "server" side of things, i.e. your node application.

> but in my react application, in src\model\TestUSB.ts I have code that looks like this:
>
> ```
> let HID: any;
> if (process.env.REACT_APP_MODE === 'electron') {
>     console.log(`REQUIRING node-hid`); [...]

Sounds reasonable to me because this is the electron client side of things and AFAIU (asking @todbot ) `node-hid` does not opperate in that realm. What you should do is to use the atop option that works for you and combine it with the `ipcMain` method from electron (cf. https://electronjs.org/docs/api/ipc-main).

I haven't tested it yet but I will in my personal project tomorrow (hopefully) and can post back to this thread with my code.

Hope this helps!

tahesse, 2020-01-22:
Readnig through https://electronjs.org/docs/tutorial/application-architecture#main-and-renderer-processes:
> In normal browsers, web pages usually run in a sandboxed environment and are not allowed access to native resources. Electron users, however, have the power to use Node.js APIs in web pages allowing lower level operating system interactions.

My suggestion seems to be a mere workaround.

It might be worthwhile to follow https://electronjs.org/docs/tutorial/using-native-node-modules.

todbot, 2020-01-22:
Hi @tahesse, 
You can use `node-hid` (or any native-module) in the Electron renderer process as well as in the Electron main process.  Electron will transparently use its IPC remote mechanisms to allow this. 

For an example, see https://github.com/todbot/electron-hid-toy. 

In renderer, you do have to make sure you open up the `BrowserWindow` from main with `nodeIntegration:true`.   

You also need to make sure that the native modules have been recompiled for Electron.  By default `node-hid` ships with Electron prebuilds for Mac, Windows, and Ubuntu/Debian, so you should be able to do `electron-rebuild` to get it to download the correct native module for your situation.