node-hid/node-hid #320: Problem connecting to ergometer

tijmenvangulik, 2019-07-22:
I tried to connect to a concept 2 ergometer. On the mac I can connect but the device never returns data. There is no error message. On the PC (running as virtual machine on the same mac) the device returns data with the same source code.  But after closing the connection I can not reconnect again.

Any idea how to fix this?

versions
- node-hid 0.7.9
- node 2.6.8

windows reconnect error
\node_modules\node-hid\nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^
Error: cannot open device with path \\?\hid#vid_17a4&pid_0001#8&229f0d25&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
    at new HID (DemoNode\node_modules\node-hid\nodehid.js:49:17)
    at Object.<anonymous> (C:\Users\tijmen\Documents\ErgometerJsUsb\DemoNode\dist\app.js:12:18)
    at Module._compile (internal/modules/cjs/loader.js:816:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
    at Module.load (internal/modules/cjs/loader.js:685:32)
    at Function.Module._load (internal/modules/cjs/loader.js:620:12)
    at Function.Module.runMain (internal/modules/cjs/loader.js:877:12)
    at internal/main/run_main_module.js:21:11

source code in typescript: 
`
import * as HID from "node-hid";

var devices = HID.devices();
var deviceInfo = devices.find( function(d) {
    return d.vendorId==6052;
});
if( deviceInfo ) {
  var device = new HID.HID( deviceInfo.path );
  
  device.on('data', function(data) {
      if (data && data.length>0)
        console.log("got data:", data.toString('hex') );
});
// 
  device.on("error", function(err) {
      console.error(err)
  });
  device.setNonBlocking(true);
  device.resume();

 var written= device.write([0x02,0xf1, 0x81, 0x81, 0xf2]);
  console.log("written "+written);
  
console.log("Waiting 10 seconds for data from c2");
setTimeout( function() {
    console.log("Done");
    device.close();
}, 10000);
  // ... use device
}
`

todbot, 2019-07-22:
What is the HID descriptor for this device?  

There could be two issues:

1. On Windows, try prepending a '0x00' to your `device.write()` array.  e.g. 
```
if(os.platform === 'win32') {
  buffer.unshift(0);  // prepend throwaway byte
}
```
(This is a current deficiency in the underlying C-library)

2. The other issue may be the HID report size needs to be set to the size of the REPORT_COUNT value in the descriptor.

tijmenvangulik, 2019-07-22:
Thank you for the quick reply. I can not yet find a way to set the HID report size in the api. On my mac I can get the following info on the device. 

interface:-1
manufacturer:"Concept2"
path:"IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/Concept2 Performance Monitor 3 (PM3)@14100000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14100000,0"
product:"Concept2 Performance Monitor 3 (PM3)"
productId:1
release:256
serialNumber:"400122800"
usage:165
usagePage:65440
vendorId:6052

 "IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/Concept2 Performance Monitor 3 (PM3)@14100000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14100000,0"

todbot, 2019-07-22:
There is no explicit way to set the report size: it's determined by the size of the array you pass in to `device.write()`.

Sorry, by HID descriptor I mean the expected data structure and "API" that defines what information is sent and received by the device.  For example, how did you know to send `[0x02,0xf1, 0x81, 0x81, 0xf2]`?

tijmenvangulik, 2019-07-22:
Thank you for looking into this problem.
I know what to send because I have on my mac an hid-api library which makes use of the qt hid-api library.  This one works very similar to the node version, and this one can communicate correctly. In node I send the same commands. (I also made sure that I configured it the same way) The other reason I know what I am sending is correct is that the same node app on windows 10 returns the correct results. (except from the connection problems, but that is minor compared to the mac version)

todbot, 2019-07-22:
Sorry I must have misspoke.  I mean, what make you think sending the particular byte sequence `[0x02,0xf1, 0x81, 0x81, 0xf2]` to a "concept 2 ergometer" does anything useful?

tijmenvangulik, 2019-07-23:
It is the get status command. It should return according to the specs
 0x02,0xF0, 0x00, 0xFD, 0x01, 0x80, 0x01, 0x01, 0x81, 0xF2
(On windows it actually does return this value)

todbot, 2019-07-23:
Yes, but how do you know it's the "get status command"?  Are you referencing some API documentation some where?  What specs?

tijmenvangulik, 2019-07-23:
Yes there are specs in the documentation when you download the software development kit.

https://www.concept2.com/service/software/software-development-kit <https://www.concept2.com/service/software/software-development-kit>

 And there is an example library hid library

https://github.com/ergarcade/c2proxy/tree/dev <https://github.com/ergarcade/c2proxy/tree/dev>

https://github.com/droogmic/Py3Row <https://github.com/droogmic/Py3Row>

some details from the specs

Specifically, the PM3 enumerates itself as a Human Interface Device (HID) with a control endpoint and two interrupt endpoints (IN/OUT).

Table 15 - PM3/PM4 USB Definitions

Parameter

Description

Bus Specification

USB 1.10

Bus Speed

Full-speed (12 Mbits/sec)

Control Endpoint Max Pkt Size

8 bytes

Device Description

Bus powered (98 mA max), 1 interface configuration (0)

Interface Description

Human Interface Device (HID)

Manufacturer string

“Concept2”

Product string

“Concept2 Performance Monitor 3 (PM3)” or “Concept2 Performance Monitor 4 (PM4)”

Endpoints

IN: Interrupt/EP3/polling rate: 8 msec. OUT: Interrupt/EP4/polling rate: 4 msec.

Reports

ID #1 – 20 bytes + 1 byte report ID ID #2 – 120 bytes + 1 byte report ID ID #4 – 62 bytes + 1 byte report ID 

Network and Transport Layers

The CSAFE protocol provides both the network and transport layer functionality over USB. The packetization, integrity checking, and node addressing is supported by the protocol. The maximum frame size has been increased to 96 bytes from the 32 bytes specified for the asynchronous serial interface.

todbot, 2019-07-23:
Ahh perfect.  So that last bit about the reports and their size in bytes is the important part.  When it says "ID 2 - 120 bytes + 1 byte report ID".  

This means you need to send 121 bytes to the device and the first byte needs to be `0x02`.  I assume for the sequence you are sending `[0x02,0xf1, 0x81, 0x81, 0xf2]`, you are sending on reportId 2.  

So I would try creating a 121-byte Array, and then setting the first five bytes to the sequence above.

tijmenvangulik, 2019-07-23:
Ah thank you that is the trick. I had to send a complete array of 121 bytes and not only the message. Thank you for your help now it works. Great support, I just gave node-hid an extra star.

Tijmen

todbot, 2019-07-25:
Great thanks!

closed 2019-07-25 by todbot