node-hid/node-hid #321: Path is empty on Mac after updating node-hid

PastaJ36, 2019-08-02:
I've seen the other issue about the path name too long #169 , but I'm not sure if this is the same.

I have an electron application that I build with electron-builder (21.2.0) and node-hid (0.7.9). Before I package it and run electron it works just fine, but after packaging the path variable is all empty. It worked before using node-hid 0.7.3 and seems te be fine on windows too. 

Before packaging:
[image]
After packaging:
[image]

Any thoughts where to start looking?

PastaJ36, 2019-08-02:
Can reproduce with quickstart + electron-builder + node-hid. Repo here: https://github.com/PastaJ36/node-hid-empty-path

todbot, 2019-08-02:
What version of MacOS are you using?

PastaJ36, 2019-08-03:
Version 10.14.3

PastaJ36, 2019-08-03:
Updated MacOs to 10.14.6, same thing. It works with electron-builder 21.1.1, but break with 21.1.2. That electron-builder release fixes some mac bugs, but I'm having a hard time figuring out how it's related (https://github.com/electron-userland/electron-builder/releases/tag/v21.1.2)

todbot, 2019-08-03:
Hi @PastaJ36,
I tried the repo you reference, but it has no use of `node-hid` in it. 
But if I add the following to `main.js` and `renderer.js`:
```js
const HID = require('node-hid')
console.log(HID.devices())
```
I verify that `node-hid` works and prints out paths in development mode in both main and renderer process, but when packaged it works but the `path` parameter is empty.

I'm not sure what could be causing this very weird error.  But if you found that `electron-builder` 21.1.1 works but 21.1.2 doesn't, at least we have something to go on.

I suspect that since Mac HID path strings are very long, perhaps it has something to do with it hitting some max string length? Perhaps it's a 'security feature' of Chromium or electron-builder or how Mac builds are configured. For instance, maybe it was this patch: https://github.com/electron-userland/electron-builder/commit/c8c37f20c624e5d73298d6bfa02660bb03ccbcea

todbot, 2019-08-03:
Verified that because the default Mac build style is now `hardenedRuntime` by default, `node-hid` does not get paths.  To get pre-`electron-builder@21.1.2` behavior, add this to your `package.json` "build" key:
```js
  "build": {
      "mac": {
          "hardenedRuntime": false
      }
  }
```

PastaJ36, 2019-08-03:
Thanks for taking the time to investigate, I can verify it works too. Do you think it is because the Mac Hardened Runtime will not allow hid-api to get the paths? If that is so it might become troublesome to release apps in the future, because it will be a requirement for notarization: https://developer.apple.com/documentation/security/hardened_runtime_entitlements.

todbot, 2019-08-03:
It looks like a work-around (as hinted at on [Electron's"Mac App Store Submission Guide"](https://electronjs.org/docs/tutorial/mac-app-store-submission-guide) page) is to sign the native module shared library.  

For instance, with the repo we've been playing with, I've found that this works:
```
$ npm install -g electron-osx-sign
$ cp node_modules/node-hid/build/Release/HID.node dist/mac/electron-quick-start.app/Contents/MacOS
$ electron-osx-sign dist/mac/electron-quick-start.app  dist/mac/electron-quick-start.app/Contents/MacOS/HID.node
$ open dist/mac/electron-quick-start.app
``` 
and the paths are back. (This overrides the same file in "app.asar" because I believe shared libraries in the same dir as the executable are searched before ones exploded out by the asar expander)

I suspect as this becomes more of an issue, there will be an electron-builder + webpack solution that everyone will start using.

krpiatkowski, 2019-09-25:
@todbot 
What your doing essencially is resigning the app to not use hardened runtime anymore if you set the --hardenedRuntime flag it will stop working again.

I have used the last few days of my life to figure out the solution is to set "com.apple.security.cs.allow-dyld-environment-variables" as a entitlement
I have no idea why this works, but i have a app which has been notarized and is running on catalina afterwards.

If anyone of the developers would chip in, it would be nice.

gniezen, 2019-09-25:
I can confirm that @krpiatkowski 's fix works for me as well. For me it presented itself as a `cannot open device with vendor id 0x1a79 and product id 0x6210` error, so it took a while to figure out that it was the missing entitlement that was at fault.

todbot, 2019-09-26:
Ah, thanks @krpiatkowski & @gniezen, this is useful info for anyone using devices in Electron.  In a perfect world, `electron-builder` should handle this.

PastaJ36, 2019-12-27:
Small update after going through the full notarisation. I've had to add `com.apple.security.cs.disable-library-validation` to the list of entitlements too, otherwise the path was still empty. It feels a little bad to just add these without fully understanding what they do, but I have no time to dive deeper into mac builds. 

So in total my plist looks like:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
  <dict>
    <key>com.apple.security.cs.allow-unsigned-executable-memory</key><true/>
    <key>com.apple.security.cs.allow-dyld-environment-variables</key><true/>
    <key>com.apple.security.cs.disable-library-validation</key><true/>
  </dict>
</plist>
```