node-hid/node-hid #322: Cant connect to device on Mac .. works on Windows

warrick-eosny, 2019-08-14:
I have a device that I'm able to connect to on windows, but when I run the same code on Mac it doesn't seem to work.

I've confirmed that I can connect to other devices like the Blink1 on the same Mac. 

When listing the devices, I do see the device. 

```
  { vendorId: 4292,
    productId: 35535,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/USB2.0 Hub@14300000/AppleUSB20Hub@14300000/AppleUSB20HubPort@14310000/Metro@14310000/Metro@0/IOUSBHostHIDDevice@14310000,0',
    serialNumber: '0123456789ABCDEF',
    manufacturer: 'Silicon Labs',
--
    serialNumber: '0123456789ABCDEF',
    manufacturer: 'Silicon Labs',
    product: 'Metro',
    release: 256,
    interface: -1,
    usagePage: 61904,
    usage: 1 }
```

However when I run the following:

```
Last login: Wed Aug 14 14:22:04 on ttys007
Warricks-MBP:metro-web-electron-debugger wfitzgerald$ pwd
/Users/wfitzgerald/dev/metro-api/examples/metro-web-electron-debugger
Warricks-MBP:metro-web-electron-debugger wfitzgerald$ cd ..
Warricks-MBP:examples wfitzgerald$ owd
-bash: owd: command not found
Warricks-MBP:examples wfitzgerald$ pwd
/Users/wfitzgerald/dev/metro-api/examples
Warricks-MBP:examples wfitzgerald$ cd
Warricks-MBP:~ wfitzgerald$ cd dev/
Warricks-MBP:dev wfitzgerald$ mkdir hid
Warricks-MBP:dev wfitzgerald$ cd his
-bash: cd: his: No such file or directory
Warricks-MBP:dev wfitzgerald$ cd hid
Warricks-MBP:hid wfitzgerald$ ls
Warricks-MBP:hid wfitzgerald$ npm init
This utility will walk you through creating a package.json file.
It only covers the most common items, and tries to guess sensible defaults.

See `npm help json` for definitive documentation on these fields
and exactly what they do.

Use `npm install <pkg>` afterwards to install a package and
save it as a dependency in the package.json file.

Press ^C at any time to quit.
package name: (hid) 
version: (1.0.0) 
description: 
entry point: (index.js) 
test command: 
git repository: 
keywords: 
author: 
license: (ISC) 
About to write to /Users/wfitzgerald/dev/hid/package.json:

{
  "name": "hid",
  "version": "1.0.0",
  "description": "",
  "main": "index.js",
  "scripts": {
    "test": "echo \"Error: no test specified\" && exit 1"
[... 622 lines cut ...]
// var d = new HID.HID(4292,35535); // This is my Blink1 device and this works fine. 
var d = new HID.HID(10168,493);
console.log(d);

```

I get this error:

```
/Users/wfitzgerald/dev/hid/node_modules/node-hid/nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0x27b8 and product id 0x1ed
    at new HID (/Users/wfitzgerald/dev/hid/node_modules/node-hid/nodehid.js:49:17)
    at Object.<anonymous> (/Users/wfitzgerald/dev/hid/test.js:8:9)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
    at startup (internal/bootstrap/node.js:279:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:752:3)
```

todbot, 2019-08-14:
Is this device trying to emulate a mouse or keyboard?  The OS will not let you grab those, as a rule

warrick-eosny, 2019-08-14:
It is a standard HID device. But I don't believe it's trying to emulate a Keyboard or Mouse. But my knowledge of USB is very limited. Is there a way to test / see if the OS considers it to be a keyboard or Mouse. 

The device is a u2f key, similar to this one -  https://solokeys.com/

warrick-eosny, 2019-08-14:
Ah hold up a minute, something this magically started working ... I'm very confused .. sorry

closed 2019-08-15 by warrick-eosny