node-hid/node-hid #323: get data from barcode scanner

freshman2008, 2019-08-15:
Use barcode scanner,
  var device = new HID.HID( deviceInfo.path );
  // ... use device
  device.on("data", function(data) {
   console.log("Raw", data);
  });
I scanner an barcode(actual value: 6957842208856)
but in callback function, I get following log
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 23 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 26 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 22 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 24 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 25 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 21 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 1f 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 1f 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 27 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 25 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 25 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 22 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 23 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 00 00 28 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>
<Buffer 03 00 00 00 00 00 00 00>
<Buffer 03 00 0d 00 00 00 00 00>
<Buffer 03 00 00 00 00 00 00 00>
<Buffer 00 00 00 00 00 00 00 00>

how can I transfer these data to the valid data(6957842208856)?

todbot, 2019-08-15:
It sounds like you are asking sort of two questions here: 
1. How to convert Node.js Buffer to something else (Array, etc)
2. What is the output data format of the barcode scanner.

I can't help you with the second question. As to the first, there are many places discussing conversion of Buffers to something else. The easiest might be just to iterate over it (since it acts like an array. You can build up your own array from this:
```js
// make up a fake buffer to test with
var buf = new Buffer.from([0, 23, 42, 55, 0, 9, 88, 255, 5])
var arr = []; // array to be created
// inspect each element
for( var i=0; i<buf.length; i++) { 
  console.log( buf[i] )
  arr.push(buf[i]) // add to the array
}
console.log(arr)
```

freshman2008, 2019-08-19:
Thanks todbot first.
Actually, the second question is what I want to know.
I have no idea about the data format of the barcode scanner.
I search on search engine, can't find the right result so far.
I only know that barcode scanner is work like a keyboard, so the data buffer is a keydown and a keyup data array, but how to map it to a real key(like key num1, num2).

eminarslan, 2019-11-18:
@freshman2008 can you share your code?

closed 2019-11-23 by todbot

raghunila, 2021-12-15:
interested in your answer? could please share.