node-hid/node-hid #326: getting error while installing a package which uses node-hid

jigneshMitel, 2019-09-06:
`> node-hid@0.7.9 install /usr/lib/node_modules/tezster/node_modules/node-hid
> prebuild-install || node-gyp rebuild

prebuild-install WARN install EACCES: permission denied, access '/home/jignesh/.npm'
gyp WARN EACCES user "root" does not have permission to access the dev dir "/home/jignesh/.node-gyp/10.16.3"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/lib/node_modules/tezster/node_modules/node-hid/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/tezster/node_modules/node-hid/.node-gyp'
gyp ERR! System Linux 4.15.0-1040-gcp
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/tezster/node_modules/node-hid
gyp ERR! node -v v10.16.3
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-hid@0.7.9 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.7.9 install script.`

I'm using **node : v10.16.3**
**npm : 6.9.0**
**ubuntu 18.04 LTS**

todbot, 2019-09-06:
Hi,
This is not a problem with `node-hid` but rather using node and npm with `sudo`.  See: https://medium.com/@ExplosionPills/dont-use-sudo-with-npm-still-66e609f5f92

closed 2019-09-06 by todbot

jigneshMitel, 2019-09-11:
@todbot thanks for pointing in the right direction but my package (which has node-hid dependency) needs to be installed globally and global install won't work without root user's permission. Can you suggest some workaround?

todbot, 2019-09-11:
I've only used node & npm in exclusive non-root situations. 
When you were installing, did you use 'sudo' or run as the 'root' user or did you just do "npm install node-hid" as user 'jignesh'?