node-hid/node-hid #341: Uncaught Error: cannot open device with path

GanZhiXiong, 2019-09-24:
[image]

Uncaught Error: cannot open device with path IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/USB2.0 Hub@14300000/AppleUSB20Hub@14300000/AppleUSB20HubPort@14310000/HELLOKEY-30A31F8ED8F3BFCE@14310000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14310000,0
    at new HID (/Users/ganzhixiong/Documents/electron/github/hellokey-desktop/node_modules/node-hid/nodehid.js:49)

todbot, 2019-09-24:
If the device is a keyboard (or acts like a keyboard), the OS will not allow you to open it for security reasons.

closed 2019-09-24 by todbot

GanZhiXiong, 2019-09-24:
This is a usb HID devices, using HIDSharp can open the equipment (https://www.zer7.com/software/hidsharp).
Is there no other way?

GanZhiXiong, 2019-09-24:
Looking forward to your reply! Thanks 😀

todbot, 2019-09-24:
Yes, but what kind of a HID device is it?  Do you have details on it?

GanZhiXiong, 2019-09-25:
[image]

I can only provide software to get device information, because I do not understand the hardware, other may not be able to tell!

GanZhiXiong, 2019-09-25:
Equipment information is as follows：

Information for device HELLOKEY-30A31F8ED8F3BFCE (VID=0x1A86 PID=0xE0E0):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0004
Current configuration value: 0x01
Number of open pipes: 2

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x1A86	idVendor
0xE0E0	idProduct
0x0000	bcdDevice
0x01	iManufacturer
0x02	iProduct
0x03	iSerialNumber
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x04	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0x23	bMaxPower      (70 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x01	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x05	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0100	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0065	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x18	bInterval         (24 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x18	bInterval         (24 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  
0x01   0x0000  Request failed with 0x0000001F
0x02   0x0000  Request failed with 0x0000001F
0x03   0x0000  Request failed with 0x0000001F
0x04   0x0000  Request failed with 0x0000001F
0x05   0x0000  Request failed with 0x0000001F

------------------------------

Connection path for device: 
Intel(R) 82801FB/FBM USB Universal Host Controller - 2658
Root Hub
Generic USB Hub
HELLOKEY-30A31F8ED8F3BFCE (VID=0x1A86 PID=0xE0E0) Port: 1

Running on: Windows 10 or greater (Build Version 18362)

Brought to you by TDD v2.12.0, Aug  2 2019, 16:35:05

GanZhiXiong, 2019-09-25:
@todbot   I have been puzzled by this question for a long time. I look forward to your reply very much. Thank you!

I am a developer from China. We may have different time zones, so I will be very happy to receive your reply and reply to you immediately. 😂

GanZhiXiong, 2019-09-26:
@tobyjoe   @todbot @DracoBlue  @mdeltito

GanZhiXiong, 2019-09-27:
@todbot  Please help me to have a look when you are free. Thank

timothyparez, 2019-09-29:
@todbot I can confirm this is indeed an HID device that has the ability to act as a keyboard.  
Aside from that it also supports communication via HID for configuration.

Using C# and the [HidSharp](https://www.zer7.com/software/hidsharp) library the device
can be controlled without any issues on MacOS, Windows and Linux.

So it would seem this issue is either specific to node-hid or an attempt is made to open the wrong device.

GanZhiXiong, 2019-09-29:
In Windows calls HID. Devices (), one device corresponds to two device information and can be turned on

``` C#
  {
    vendorId: 6790,
    productId: 57568,
    path: '\\\\?\\hid#vid_1a86&pid_e0e0&col02#7&83dd5c8&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',      
    serialNumber: '30A31F8ED8F3BFCE',
    manufacturer: 'TJT-USB',
    product: 'HELLOKEY-30A31F8ED8F3BFCE',
    release: 0,
    interface: -1,
    usagePage: 1,
    usage: 6
  },
  {
    vendorId: 6790,
    productId: 57568,
    path: '\\\\?\\hid#vid_1a86&pid_e0e0&col01#7&83dd5c8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: '30A31F8ED8F3BFCE',
    manufacturer: 'TJT-USB',
    product: 'HELLOKEY-30A31F8ED8F3BFCE',
    release: 0,
    interface: -1,
    usagePage: 65280,
    usage: 1
  },
```

denyncrawford, 2020-12-11:
> 
> 
> @todbot I can confirm this is indeed an HID device that has the ability to act as a keyboard.
> Aside from that it also supports communication via HID for configuration.
> 
> Using C# and the [HidSharp](https://www.zer7.com/software/hidsharp) library the device
> can be controlled without any issues on MacOS, Windows and Linux.
> 
> Basically with HidSharp you can see to ports, the first one is the keyboard
> the second one is the HID interface which can be used to configure the device.
> 
> I am guessing only the first port is detected by node-hid and of course
> if you try to open that one the operating system will prevent you from doing that.
> 
> Is there a way to specify which port/endpoint to open?
> 
> ```
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x82  EP 2 IN
>         bmAttributes            3
>           Transfer Type            Interrupt
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0040  1x 64 bytes
>         bInterval              24
>       Endpoint Descriptor:
>         bLength                 7
>         bDescriptorType         5
>         bEndpointAddress     0x02  EP 2 OUT
>         bmAttributes            3
>           Transfer Type            Interrupt
>           Synch Type               None
>           Usage Type               Data
>         wMaxPacketSize     0x0040  1x 64 bytes
>         bInterval              24
> ```

Hi, did you found a soution? I'm running a barcode scanner that acts like a keyboard and it is not configurable or switchable to work just with HID (as @todbot says on multiple answers).

todbot, 2020-12-11:
@denyncrawford in the case of a device that is a keyboard but also has other usages or non-keyboard functions, I would do a `HID.devices()` and find the usagePage / usage combo of your device that your HidSharp code is using to access it.

And example of how to do this is https://github.com/node-hid/node-hid/blob/master/src/test-teensyrawhid.js, where the Teensy shows up as multiple HID devices, only one of which is valid for communication.

todbot, 2020-12-11:
But if you want to read any keyboard in Node/Electron, you can use standard keyboard events, no need for a special library like `node-hid`.  The problem (and the "feature" of things like these HID barcode scanners) is they are indistinguishable from keyboards to the OS.

denyncrawford, 2020-12-12:
@todbot thank you for the answer. You're right, you don't need to treat it as nothing more than a keyboard. Looking a bit I found [this module](https://github.com/hadeeb/simple-barcode-scanner) that allows to classify if the keyboard is a scanner by it's write latency, and then use it with events, it is really simple. Actually, the problem is that it is impossible to know from which device the event is emitted (directly from the process) and sometimes it is important, especially when working in applications that require multiple devices. 

**💡 So, I had an idea**:

 Some devices have the option to attach a prefix string to the barcode, so why not use that to classify the device that is being used? Well, I did it on [this PR](https://github.com/hadeeb/simple-barcode-scanner/pull/13).
 Now the "**instance**" can sort the events if the barcode includes the `options.devicePrefix` value (if no prefix is specified, it can be used globally as usual).

I think the repository is not constantly maintained, so I don't know if they will accept  the PR, but it works perfectly and you can use it by [coning the my fork](https://github.com/denyncrawford/simple-barcode-scanner/tree/device-instance) or even using it with Deno, script:src, etc. since it is a simple window compatible function.

Hope this helps whoever needs it, regards.