node-hid/node-hid #345: can not open device in docker

misland, 2019-10-23:
@todbot hi,I am testing this library in docker(node:10.16.3-jessie) and my OS is Ubuntu 18.04,the code is quite simple as follows:
'use strict';
const HID = require('node-hid'); 
let devices = HID.devices();
console.log(devices);
let device = new HID.HID(1155, 14129);
when executed in docker,it can list the devices' info connected to the pc but it just cann't open the device by VID and PID.

misland, 2019-10-23:
here is the error message:
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0x483 and product id 0x3731
    at new HID (/or/app/av/audio/api/node_modules/node-hid/nodehid.js:49:17)
    at Object.<anonymous> (/or/app/av/audio/api/test.js:25:14)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:831:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)

todbot, 2019-10-23:
Two questions to ask when having problems opening the device:
- Do you have udev rules in place or are running as 'root'?
- What kind of device is this?  If it's a mouse or keyboard (or acts like one), you cannot open it as the OS prevents non-OS access

misland, 2019-10-24:
@todbot 
well,it's a sad story >_<,I do didn't set the rules in the docker container for the device,but when I try to do this,I found this
[image]
so,I think docker can not use the hid device really. 
maybe I can close issue now.

closed 2019-10-25 by misland