node-hid/node-hid #348: macOS cannot open device with path IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/USB2.0 Hub@14300000/AppleUSB20Hub@14300000/AppleUSB20HubPort@14310000/HELLOKEY-8F3A4DEAFF1A0D6A@14310000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14310000,0

GanZhiXiong, 2019-11-07:
[image]
[image]

@todbot  Using 'node-hid' can open the device on Windows and Linux, but it cannot be opened on macOS. Before this problem, we submitted an issue in #341, but you turned it off. Your reply to me is as follows:
> If the device is a keyboard (or acts like a keyboard), the OS will not allow you to open it for security reasons.

 I searched in Issues, and quite a few people have this problem? Now I still want to know how to solve this problem? How do I open a device on macOS?

todbot, 2019-11-07:
Is the device a keyboard-like device?

GanZhiXiong, 2019-11-07:
First of all, thank you very much for the package 'node-hid'. With 'node-hid', we have finished the programs on Windows and Linux. Now we are left with macOS, That all. Thank you very much 🙏 🙏

GanZhiXiong, 2019-11-07:
The device is similar to Yubico, and can automatically input characters by touching the device

todbot, 2019-11-07:
It sounds like it is acting like a keyboard. I that case, you will not be able to open it because the OS prevents it for security reasons. Windows also prevents this on the most recent versions, I believe. Otherwise malicious programs could open the keyboard and read it as a user types in a password.

GanZhiXiong, 2019-11-07:
> It sounds like it is acting like a keyboard. I that case, you will not be able to open it because the OS prevents it for security reasons. Windows also prevents this on the most recent versions, I [...]

Hi, todbot, is there no other way?

todbot, 2019-11-07:
If your device is like a YubiKey, I’d recommend changing its protocol to follow the U2F protocol https://fidoalliance.org/specs/u2f-specs-1.0-bt-nfc-id-amendment/fido-u2f-hid-protocol.html

todbot, 2019-11-07:
Also, if you’re just trying to read “keypresses” from the device and your app is command line, you can use process.openStdin()

GanZhiXiong, 2019-11-07:
> Also, if you’re just trying to read “keypresses” from the device and your app is command line, you can use process.openStdin().
> 
> Or it’s it’s an Electron app, you can use standard browser keypress events

Both reading and writing.
I'll try objective-c when I have time. If objective-c can open the device, it should be a 'node-hid' problem.

todbot, 2019-11-07:
`node-hid` is just a thin wrapper around [hidapi](https://github.com/libusb/hidapi), which the macOS portion is written in ObjectiveC. You can see the [macOS part here](https://github.com/libusb/hidapi/blob/master/mac/hid.c)

GanZhiXiong, 2019-11-08:
@todbot   Thank you for your information😀

cahitihac, 2019-11-12:
Hi @todbot ,

I have a similar issue with a HID device (Jabra Speak 710 speakerphone) that's happening only on MacOS but works just fine on Windows.

This is the error I'm getting when trying to write to the device using **[electron-hid-toy](https://github.com/todbot/electron-hid-toy)** app:

[image]

The idea that a device might be acting like a mice/keyboard so the OS prevents any attempt to write to the device makes sense in my case.

So my question is how do I figure out or verify that my device is acting like a keyboard/mice?

Is there any or an easy way to do that?

todbot, 2019-11-12:
Hi @cahitihac,
You should be able to get the properties 'usagePage' and 'usage' from the `HID.devices()` list.  From that you can compare it to the pre-define [HID Usage Tables](https://www.usb.org/document-library/hid-usage-tables-112).  Many of those usagePages are grabbed by the OS.

cahitihac, 2019-11-13:
Thanks @todbot .

Looks like my issue is not because of the device is acting like a keyboard.

Jabra Speak 710:
`{
   "vendorId":2830,
   "productId":9333,   "path":"IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/Jabra Speak 710@14100000/IOUSBHostInterface@3/IOUSBHostHIDDevice@14100000,3",
   "serialNumber":"70BF9204AA1E",
   "manufacturer":"",
   "product":"Jabra Speak 710",
   "release":296,
   "interface":-1,
   "usagePage":11, //Telephony page
   "usage":5 //Headset usage
}`

Mouse:
`{
   "vendorId":1452,
   "productId":657,   "path":"IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@15,4/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Trackpad / Boot/AppleHSSPIHIDDriver",
   "serialNumber":"",
   "manufacturer":"",
   "product":"Apple Internal Keyboard / Trackpad",
   "release":369,
   "interface":-1,
   "usagePage":1, //Generic desktop controls
   "usage":2 //mouse usage
}`

Keyboard:
`{
   "vendorId":1452,
   "productId":657,	"path":"IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/SPI1@15,4/AppleIntelLpssGspi@1/AppleIntelLpssSpiController@1/AppleIntelLpssSpiDevice@0/AppleHSSPIController/Apple Internal Keyboard / Trackpad/Keyboard / Boot/AppleHSSPIHIDDriver",
   "serialNumber":"",
   "manufacturer":"",
   "product":"Apple Internal Keyboard / Trackpad",
   "release":369,
   "interface":-1,
   "usagePage":1, //Generic desktop controls
   "usage":6 //keyboard usage
}`

timothyparez, 2019-11-14:
> It sounds like it is acting like a keyboard. I that case, you will not be able to open it because the OS prevents it for security reasons. Windows also prevents this on the most recent versions, I [...]

There are basically two devices embedded into it.
One is a keyboard, which he does not need to 'open'
The other is an HID device used to configure the 'keyboard'.

todbot, 2019-11-14:
Hi @timothyparez,
Are you referring to the device that @cahitihac is using?  It doesn't seem like it.

@cahitihac, those three devices you list (headset, mouse, keyboard) I believe will all be claimed by the OS.

timothyparez, 2019-11-14:
@todbot No sorry, I was referring to the device GanZhiXiong is using.  
I wanted to make clear that at no point he is trying to open the keyboard device,
the keyboard is completely separate from the HID device used for configuration.

cahitihac, 2019-11-14:
> Hi @timothyparez,
> Are you referring to the device that @cahitihac is using? It doesn't seem like it.
> 
> @cahitihac, those three devices you list (headset, mouse, keyboard) I believe will all be claimed by the OS.

I understand why mouse and keyboard are claimed by the OS. But why the headset?

I have other devices produced by the same vendor, for example  **Jabra Engage 65**. And they just work. I have no problem with those devices. **Jabra Speak 710** is the only one that doesn't work.

todbot, 2019-11-14:
@GanZhiXiong or @timothyparez, do you know the `usagePage` and `usage` properties for the device to which you're referring?

@cahitihac, I'm not sure why it would be grabbing it (if it is).  Maybe because it has volume up/down/mute abilities and that's considered a system-level action?

The other possibility is that it's a compound USB device with the same VID & PID but different usagePages and when you call `HID.open(VID,PID)` it's attempting to open the wrong one.  To prevent this, you need to filter by `usagePage` and `usage`.  

To see an example of how to filter by usage, check out [test-teensyrawhid.js](https://github.com/node-hid/node-hid/blob/master/src/test-teensyrawhid.js).  The Teensy RawHID device has a similar problem where it exposes two devices, but with different usages.

cahitihac, 2019-11-14:
Yes, that's right.

I've encountered the same problem on Windows then I filtered devices by `usagePage` and `usage`. I'm using the same filter for MacOS too although on MacOS there's only one `usagePage` and one `usage` for the same device.

I still have no idea why I'm seeing this issue with only this specific device but not with other devices. Nothing is different in terms of the code base or the way that I manage devices using `node-hid`.

GanZhiXiong, 2019-11-15:
@todbot   In macOS calls HID.Devices(),  MacOS there's only one usagePage and one usage for the same device, macOS Device cannot be open.

But you can get two usagePage and two usage on the same device under Windows, and can open device.
 ```
This is the device information that you got under Windows

{
    vendorId: 6790,
    productId: 57568,
    path: '\\\\?\\hid#vid_1a86&pid_e0e0&col02#7&83dd5c8&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd',      
    serialNumber: '30A31F8ED8F3BFCE',
    manufacturer: 'TJT-USB',
    product: 'test-30A31F8ED8F3BFCE',
    release: 0,
    interface: -1,
    usagePage: 1,
    usage: 6
  },
  {
    vendorId: 6790,
    productId: 57568,
    path: '\\\\?\\hid#vid_1a86&pid_e0e0&col01#7&83dd5c8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: '30A31F8ED8F3BFCE',
    manufacturer: 'TJT-USB',
    product: 'test-30A31F8ED8F3BFCE',
    release: 0,
    interface: -1,
    usagePage: 65280,
    usage: 1
  },
``` 

@cahitihac  Do you have the same problem?

cahitihac, 2019-11-15:
Hi @GanZhiXiong,

I'm not sure if it's a problem, but more of a difference between OS's.

And yes I do have the same situation. For each device I have at least 3 usages, 4 for some devices, in Windows. And only one usage for every device in MacOS.

GanZhiXiong, 2019-11-15:
Hi @cahitihac  

I've done the communication between the device on Linux and Windows, except for macOS, and I think there's actually more than one usage on one device on macOS, probably limiting access to other usage for security reasons, and if there are other USES available, I think I can turn the device on

cahitihac, 2019-11-15:
I see. That makes sense.

In my case, on MacOS, I get the only usage that I need. So I don't have to worry about other usages.

GanZhiXiong, 2019-11-25:
@todbot New version 1.0.0 doesn't solve my problem either

todbot, 2019-11-25:
Hi @GanZhiXiong,
Correct, node-hid@1.0.0 only switches to latest hidapi, but the latest hidapi doesn't include the fix.  (It's still a PR and the original author of the PR hasn't responded. I'm looking into doing a new PR that addresses the issue, but I don't have a piece of hardware that I can test against)

GanZhiXiong, 2019-11-25:
Hi, @todbot, I in China, if closer I'll 😂 provides a device for you. I guess I'll have to wait. Thanks!

todbot, 2019-11-25:
Hi @GanZhiXiong,
If possible, could you use the Mac USB Prober.app and show me what your device's HID report descriptor looks like?  I may be able to create a device that duplicates the usagePage behavior you're seeing.  

In USB Prober, for instance, the blink(1) mk3 device looks like the screenshot below. But it's easier for me to read the text file created from the "Save Output" button.

You can get "USB Prober.app" by either [following the instructions here](https://superuser.com/a/814339/7614) or you can [download a zip of it here](https://github.com/node-hid/node-hid/releases/download/v1.0.0/USB.Prober-Mac.zip):

[image]

GanZhiXiong, 2019-11-25:
[test-device.txt](https://github.com/node-hid/node-hid/files/3885061/test-device.txt)

Hi, @todbot , I exported my device information as test-device.txt, looking forward to your reply! 😀😀

GanZhiXiong, 2019-11-26:
I tested it on my colleague's computer today, and found that macOS Mojave 10.14.6 could open the device, but macOS Mojave 10.14.5 could not open the device, I will try to upgrade the system when I have time

GanZhiXiong, 2019-11-26:
My upgrade to macOS Mojave 10.14.6 also failed
> One problem I forgot to report is that in macOS Mojave 10.14.6 and 10.14.5, when I insert the device, I immediately execute 'var device = hidd.hid (deviceinfo.path); Device. Close () 'can open the device, and then executing the code again cannot open the device
An error is as follows

`
Error: cannot open device with path IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/USB2.0 Hub@14300000/AppleUSB20Hub@14300000/AppleUSB20HubPort@14310000/HELLOKEY-30A31F8ED8F3BFCE@14310000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14310000,0
    at new HID (/Users/ganzhixiong/Downloads/node-hid-test/node_modules/node-hid/nodehid.js:49:17)
    at Object.<anonymous> (/Users/ganzhixiong/Downloads/node-hid-test/test.js:13:18)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:829:12)
    at startup (internal/bootstrap/node.js:283:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:622:3)
`

todbot, 2019-11-26:
Hi @GanZhiXiong I am a little unclear as to the code you're talking about regarding "immediately execute... Device. Close()".  Please provide a code snippet.

If you are using something like `node-usb-detection` to detect insertion, you will find that while the OS'es USB driver detects the device, the OS's HID driver has not had time to set it up.  You will need to insert a delay between a node-usb-detection event and seeing the device in `HID.devices()`

Similarly, due to the delays in the OS's HID system (and hidapi) taking a while to build up and tear down the infrastructure to communicate to HID devices, you cannot do something like `device = new HID.HID(vid,pid); device.close(); device = new HID.HID(vid,pid);` without some delay between the close and the open. I would try 500 msecs.

This  "open after close" problem is not well understood but seems to be part of hidapi as well as node-hid.

GanZhiXiong, 2019-11-27:
Hi, @todbot 
[image]
Above is my code

Once again, on my computer (MacBook Pro 2018) macOS Mojave 10.14.5, I can open the device by inserting the device and executing 'node test.js' immediately. When I insert the device and wait a few seconds to execute 'node test.js', I cannot open the device.

> My English is not very good, if you still don't understand, please ask😀

todbot, 2019-12-05:
Hi @GanZhiXiong,
The `hidapi` updates that allow Macs to see devices with multiple usages as unique items has been added to `node-hid` as of version v1.1.0.  This issue should be closable now. Thanks!

closed 2019-12-05 by todbot

GanZhiXiong, 2019-12-06:
HI @tobyjoe 
now macOS can see devices with multiple usages 
```
  { vendorId: 6790,
    productId: 57568,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/USB2.0 Hub@14300000/AppleUSB20Hub@14300000/AppleUSB20HubPort@14310000/HELLOKEY-8F3A4DEAFF1A0D6A@14310000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14310000,0',
    serialNumber: '8F3A4DEAFF1A0D6A',
    manufacturer: 'TJT-USB',
    product: 'Test-8F3A4DEAFF1A0D6A',
    release: 0,
    interface: 0,
    usagePage: 65280,
    usage: 1 },
  { vendorId: 6790,
    productId: 57568,
    path:
     'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/USB2.0 Hub@14300000/AppleUSB20Hub@14300000/AppleUSB20HubPort@14310000/HELLOKEY-8F3A4DEAFF1A0D6A@14310000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14310000,0',
    serialNumber: '8F3A4DEAFF1A0D6A',
    manufacturer: 'TJT-USB',
    product: 'Test-8F3A4DEAFF1A0D6A',
    release: 0,
    interface: 0,
    usagePage: 1,
    usage: 6 }
```
But I still can't open the device, I tested two macOS can open the device, I think it is my computer problem, so I leave this problem for now.

**Thank you very much for your reply 😊, I like 'node - hid 👍**