node-hid/node-hid #349: Cant connect to device on Linux .. works on Windows

travis5491811, 2019-11-07:
Hello. Similar issue as [here](https://github.com/node-hid/node-hid/issues/322) but in my case it has not "magically started working".

The project works fine on Windows 10. When i run on Linux using `node 12.13.0` using `node-hid 0.7.9` i get the following:

```
var HID = require('node-hid');
var devices = HID.devices();
console.log("hello devices", devices);

// Fairbanks scale
const vendorId = 2919;
const productId = 21854;
var foundDevice = devices.find(function(element) {
  return element.vendorId === vendorId && element.productId === productId;
});

console.log("foundDevice", foundDevice);

if (foundDevice) {
  var device = new HID.HID(vendorId,productId);

  device.on("data", function(data) {
    console.log("we got data", data);
    console.log("buffer to json", JSON.stringify(data));
  });
} else {
  console.log('the device vendorId', vendorId, ' product id', productId, 'not found');
}
```

**Console output and ERROR**
```
hello devices [
  {
    vendorId: 2919,
    productId: 21854,
    path: '/dev/hidraw4',
    serialNumber: '',
    manufacturer: 'Fairbanks Scales',
    product: 'SCB-R9000-14U Scale',
    release: 4,
    interface: 0,
    usagePage: 44576,
    usage: 499
  },
  {
    vendorId: 1133,
    productId: 49960,
    path: '/dev/hidraw0',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'USB Keyboard',
    release: 34304,
    interface: 0,
    usagePage: 3842,
    usage: 2588
  },
  {
    vendorId: 1133,
    productId: 49960,
    path: '/dev/hidraw1',
    serialNumber: '',
    manufacturer: 'Logitech',
    product: 'USB Keyboard',
    release: 34304,
    interface: 1,
    usagePage: 15370,
    usage: 25604
  },
  {
    vendorId: 2362,
    productId: 9522,
    path: '/dev/hidraw2',
    serialNumber: '',
    manufacturer: 'PixArt',
    product: 'Gaming Mouse',
    release: 256,
    interface: 0
  },
  {
    vendorId: 2362,
    productId: 9522,
    path: '/dev/hidraw3',
    serialNumber: '',
    manufacturer: 'PixArt',
    product: 'Gaming Mouse',
    release: 256,
    interface: 1,
    usagePage: 3908,
    usage: 699
  }
]
foundDevice {
  vendorId: 2919,
  productId: 21854,
  path: '/dev/hidraw4',
  serialNumber: '',
  manufacturer: 'Fairbanks Scales',
  product: 'SCB-R9000-14U Scale',
  release: 4,
  interface: 0,
  usagePage: 44576,
  usage: 499
}

/home/travis/Documents/NodeCode/HIDproject/node_modules/node-hid/nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0xb67 and product id 0x555e
    at new HID (/home/travis/Documents/NodeCode/HIDproject/node_modules/node-hid/nodehid.js:49:17)
    at Object.<anonymous> (/home/travis/Documents/NodeCode/HIDproject/index.js:15:16)
    at Module._compile (internal/modules/cjs/loader.js:956:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:973:10)
    at Module.load (internal/modules/cjs/loader.js:812:32)
    at Function.Module._load (internal/modules/cjs/loader.js:724:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1025:10)
    at internal/main/run_main_module.js:17:11
```

It looks like its looking for the hex equivalent but if i change vendorId and productId to the hex equivalent, i get no error, but it doesn't find the device.

todbot, 2019-11-07:
On Linux you will need to install udev rules for your device. Or run everything as ‘root’ (not recommend for node projects as root can screw up the npm database for all other users)

travis5491811, 2019-11-08:
Thanks. This works for fairbanks scale as amended from your docs

The device info can be found from the following:
```
$ lsusb
> Bus 001 Device 008: ID 0b67:555e Fairbanks Scales SCB-R9000

$ lsusb -vvv // has more detailed output
```

sudo vi /etc/udev/rules.d/50-usb-scale.rules
```
SUBSYSTEM=="input", GROUP="input", MODE="0666"
SUBSYSTEM=="usb", ATTRS{idVendor}=="0b67", ATTRS{idProduct}=="555e", MODE:="666", GROUP="plugdev"
KERNEL=="hidraw*", ATTRS{idVendor}=="0b67", ATTRS{idProduct}=="555e", MODE="0666", GROUP="plugdev"
```

save and then `sudo udevadm control --reload-rules && sudo udevadm trigger`

> On Linux you will need to install udev rules for your device. Or run everything as ‘root’ (not recommend for node projects as root can screw up the npm database for all other users)

closed 2019-11-08 by travis5491811