node-hid/node-hid #350: System cannot find message text for message number 0x%1 in the message file for %2

mdemaredeb, 2019-11-12:
Hi!
It's throwing the error `System cannot find message text for message number 0x%1 in the message file for %2` when I try to run HID.devices()
I'm using nwjs 0.29.1 (which uses node 9.8.0)

This is the complete error log:
[image]

If you have any idea of what is happening, it will be appreciated. Are there any places where I could find additional logs?
Thanks in advance

mdemaredeb, 2019-11-12:
After restarting the PC, the error message changed to an even more cryptic one:
`Uncaught Error: error: 317`
[image]

todbot, 2019-11-12:
This sounds like an issue with NW.js not getting a native module it can understand. Have seen docs like this: http://docs.nwjs.io/en/latest/For%20Users/Advanced/Use%20Native%20Node%20Modules/

That gets complicated, so for Electron and Node.js, I recommend [prebuild](https://github.com/prebuild/prebuild). I've not used NW.js in a while, but I heard that `prebuild` should do everything for you with a command like:
```
npm install -g prebuild
prebuild -t 0.29.1 -r node-webkit
```

todbot, 2019-11-12:
On further investigation, I was able to do this to get `node-hid` to work with NW.js:
```
cd my-nwjs-app
npm install node-hid --save
npm install -g nw-gyp
cd node_modules/node-hid
nw-gyp rebuild --target=0.42.3 --arch=x64  // or whatever NW.js version you have
cd ../..
nwjs .
```

mdemaredeb, 2019-11-12:
Thanks! It worked!

closed 2019-11-13 by mdemaredeb