node-hid/node-hid #351: Cannot read from HID device

dheerajmalraj, 2019-11-13:
Hi,

I am unable to read the data from the device. I have tried multiple API's device.read(callback) , device.readSync() and device.on('data', function(data) {} ) but there is no data received using these API's. Since in order to read the data from the device we have to write first then do a read. I was able to successfully write to the device(cross-checked with debugger write is happening) when I am trying to read the data nothing is getting returned. Following is the code that I have written. 

```
var HID = require('node-hid');
var devices = HID.devices();
console.log(devices);
HID.setDriverType("hidraw");

var device = new HID.HID(1200, 68);

// Had to set the non-blocking to true else even write was not working.
var setValue = device.setNonBlocking(true);

var config_size_command = [0x86];
let writeValue = device.write(config_size_command);
console.log("writeValue", writeValue);
// writeValue returned is 1.

//readSync
var readValue = device.readSync();
console.log("readValue", readValue[0]);

//read with callback
device.read(function (err, data) {
    console.log("in async read");
    if (data) {
        console.log("typeof data", typeof data);
        console.log("data len", data.length);
        console.log("data ", data[0].toString('hex'), data[1].toString('hex'));
    }
    if (err) {
        console.log("error ", err);
    }
});

//data - Buffer - the data read from the device
device.on('data', function(data) {
     console.log("entire data:", data);
     console.log("device config size:", data[0]);
     console.log("got data", data.toString('hex'));
});

device.on('error', function(error) {
     console.log("error",error);
});
```
**Thanks in advance for the help!**

todbot, 2019-11-14:
What kind of device is this?  e.g. What is the full output of `HID.devices()` and do you have documentation on the HID implementation of the device?  Without knowing anything about the device you're trying to connect to, it is impossible to help.

dheerajmalraj, 2019-11-15:
The device is an electronic keypad lock.
HID.devices() the device information with similar to this
sample data, not the exact one.
```
[
  {
    vendorId: XXXX,
    productId: XX,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS01@14100000/XXXX' +
      'eLock ' +
      'Device@14100000/IOUSBHostInterface@0/IOUSBHostHIDDevice@14100000,0',
    serialNumber: '',
    manufacturer: 'XXXX Inc.',
    product: 'XXXXX Device',
    release: 256,
    interface: -1,
    usagePage: 65280,
    usage: 1
  }
]
```

**Regarding the documentation on the HID Implementation of the device:-**
We have taken up the Microchip hid implementation and created our own custom hid device.
We have written our own Firmware for the device. Currently, using the node-hid library, we are trying a specific read operation that returns the size of a particular data-structure.
In order to read the information we have to write first and then do a read, we are able to put a breakpoint and see that the write is received at the device end. But we are unable to read the data.

todbot, 2019-11-15:
hi @dheerajmalraj ,

Looks like your usagePage and usage values are good.  What is also your HID report descriptor?  Are you using numbered reports?  Are you using IN/OUT reports or FEATURE reports?  What is the size in bytes of the reports you are sending & receiving?

todbot, 2019-11-15:
Hi again,
I think this may be related to this bug: https://github.com/libusb/hidapi/pull/65
I'm working on updating hidapi to include this and other patches soon.

AalapMistry, 2019-12-03:
Unable to read data from Device ?
Mydevice command is AA 02 00 00 BB
where 
AA is start  byte
02 command request   byte
00 address base high  byte
00 address base low  byte
BB stop byte

below my device information .

Info Found                                                                                        {                                                                                                   vendorId: 1155,                                                                                   productId: 22352,                                                                                 path: '\\\\?\\hid#vid_0483&pid_5750#6&d382967&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',     serialNumber: '317435563137',                                                                     manufacturer: 'STMicroelectronics',                                                               product: 'STM32 Custom Human interface',                                                          release: 512,                                                                                     interface: -1,                                                                                    usagePage: 65280,                                                                                 usage: 1                                                                                        }
--

todbot, 2019-12-03:
Hi @AalapMistry, if you're having the same issue across different HID libraries, this isn't a `node-hid` issue. I recommend you look into the HID descriptor and API description provided by your device maker.

AalapMistry, 2019-12-04:
> Hi @AalapMistry, if you're having the same issue across different HID libraries, this isn't a `node-hid` issue. I recommend you look into the HID descriptor and API description provided by your device maker.

Yes the issue was in firmware from device side now I am able to communicate with device thanks

closed 2019-12-06 by todbot