node-hid/node-hid #352: Prebuild for Node 13, please.

roddypratt, 2019-11-19:
I get this message, as there's no prebuilt Node13 binaries.

`prebuild-install WARN install No prebuilt binaries found (target=13.1.0 runtime=node arch=x64 libc= platform=win32)`

todbot, 2019-11-23:
`prebuild` and `prebuild-install` depend on `node-abi` to know which ABI versions.  `node-abi` hasn't been updated for Node 13 yet.  https://github.com/lgeiger/node-abi/blob/master/index.js#L51

When it is and when prebuild gets the new version, then node-hid can have a prebuild for it.  In the meantime, you'll need to compile the binary yourself or not use Node 13.

roddypratt, 2019-11-24:
Thanks. Looks like a PR for `node-abi` is on the way... 
https://github.com/lgeiger/node-abi/pull/74

roddypratt, 2019-12-05:
`node-abi` V2.13 has Node 13 support now. 
 https://github.com/lgeiger/node-abi/releases

todbot, 2019-12-05:
node-hid v1.1.0 (just released) has prebuilds for Node v13. thanks!

closed 2019-12-05 by todbot